kliment/Printrun #807: Where are marcos saved?

milkpirate, 2017-02-24:
Hey,

sorry that I am opening an issue but I couldnt find any info anywhere else... I am running Prontoface @ Win7x64. I found the `printrunconf.ini` in `C:\Users\%user%\` but it is not containing the privously defined macro `fan80`. It looks like so:

```
set last_window_maximized True
set last_bed_temperature 60.0
set last_extrusion 30
set last_sash_position 545
set port COM2
set baudrate 1000000
set last_file_path C:\Users\%user%\Documents\...
set recentfiles []
set last_window_width 804
set last_window_height 553
set last_temperature 190.0
set slicecommand Slic3r/slic3r.exe $s --load Slic3r/Slic3r_config_bundle.ini --output $o
set sliceoptscommand Slic3r/slic3r.exe --load Slic3r/Slic3r_config_bundle.ini --ignore-nonexistent-config
```
Where are the macros saved?

GibaHC, 2017-07-22:
Macros are saved in printrunconf.ini file ... something like this: 

```
macro SLOWDOWN
  G91
  G1 Z-0.025
  G90
```
I had issues with macros and custom buttons, and i had to edit the file to solve it.

rieders, 2017-07-22:
:)

rockstorm101, 2017-12-13:
Hi, @milkpirate, did GibaHC's answer solve your issue? If so, could you please consider closing this issue?

closed 2017-12-16 by milkpirate