kliment/Printrun #808: Making .exe file from code

soosalameh, 2017-03-01:
Hello,

I made some changes to the software and want to create a new pronterface.exe file. I am using an application called pyinstaller to do this, but am coming across a lot of issues. 

I was wondering, if anyone has done this, what program they used to make the .exe file from the python code and also, which file they chose as the one to turn into a .exe file (I was trying pronterface.py from the main printrun folder).

@kliment  Would appreciate any tips for this!

kefir-, 2017-03-01:
I don't know how it's usually done with pronterface, but https://anthony-tuininga.github.io/cx_Freeze/ is an option.

kliment, 2017-03-02:
I used py2exe, never got pyinstaller to work. The setup_win.py file and
release_windows.bat files show should be everything you need.

On 01.03.2017 23:15, soosalameh wrote:
> Hello,
>
> I made some changes to the software and want to create a new pronterface.exe file. I am using an application called pyinstaller to do this, but am coming across a lot of issues.
>
> I was [...]

closed 2017-03-13 by kliment

soosalameh, 2017-03-23:
@kliment what did you use for making the mac executable?