kliment/Printrun #810: Provide an AppImage for Linux

probonopd, 2017-03-20:
Currently there are binary downloads for macOS and Windows but not for Linux, making it harder than for the other systems to download and run Pronterface.

Providing an [AppImage](http://appimage.org/) would have, among others, these advantages:
- Works for most Linux distributions (including Ubuntu, Fedora, openSUSE, CentOS, elementaryOS, Linux Mint, and others)
- One app = one file = super simple for users: just download one AppImage file, [make it executable](http://discourse.appimage.org/t/how-to-make-an-appimage-executable/80), and run
- No unpacking or installation necessary
- No root needed
- No system libraries changed
- Just one format for all major distributions
- Works out of the box, no installation of runtimes needed (unlike Snap and Flatpak)
- Optional desktop integration with `appimaged`
- Optional binary delta updates, e.g., for continuous builds (only download the binary diff) using AppImageUpdate
- Can optionally GPG2-sign your AppImages (inside the file)
- Works on Live ISOs (unlike Flatpak)
- Can use the same AppImages when dual-booting multiple distributions

[Here is an overview](https://github.com/probonopd/AppImageKit/wiki/AppImages) of projects that are already distributing upstream-provided, official AppImages, including Ultimaker Cura, Prusa Slic3r, and FreeCAD.

probonopd, 2017-03-20:
A demonstration version can be downloaded from
https://bintray.com/probono/AppImages/Pronterface/_latestVersion#files
It should run on most 2014-ish and later distributions. Some fine-tuning may be needed. It comes with its own bundled Python installation.

This is the file that controls the AppImage conversion:
https://github.com/probonopd/AppImages/blob/master/recipes/meta/Pronterface.yml

kliment, 2018-03-25:
@probonopd the yml file you linked to is now a dead link. I'd like to make an appimage for the 2.x release that is coming up and would like your assistance with this. Would you be willing to help?

probonopd, 2018-03-28:
@kliment here is the updated link:
https://github.com/AppImage/AppImages/blob/master/recipes/Pronterface.yml

Can you use this as the basis for the newer one?

kliment, 2018-03-28:
Hmm, not easily - is it possible to make it implement the installation instructions in https://github.com/kliment/Printrun/blob/master/buildinstructions.txt (those are based on pip)?

kliment, 2018-03-28:
@rockstorm101 could you look at whether we can reuse some of the work you did for debian packaging to create an appimage?

rockstorm101, 2018-03-29:
Well, first, I don't know the first thing about AppImages :P, but yeah sure, I'll look at it.

@probonopd, please correct me if I am wrong. Looking at the '.yml' files for programs such as Geany [[0]] or GIMP [[1]], it seems like if a 'printrun-x.x.deb' file exists in advance the 'Printrun.yml' could be as simple as:
```
app: Printrun
binpatch: true

ingredients:
  package: printrun
  dist: buster
  sources: 
    - deb http://ftp.us.debian.org/debian/ buster main

script:
  - # Some corrections would surely have to be made
```
However, this approach depends on an existing package for Debian/Ubuntu, which does not seem efficient to me.

I don't think that current approach [[2]] is efficient either. It depends too much on Ubuntu packages.

I agree with @kliment, Printrun's AppImage should be built installing dependencies via pip in a manner similar to the one used for OSX [[3]].

[0]: https://github.com/AppImage/AppImages/blob/master/recipes/Geany.yml
[1]: https://github.com/AppImage/AppImages/blob/master/recipes/GIMP.yml
[2]: https://github.com/AppImage/AppImages/blob/master/recipes/Pronterface.yml
[3]: https://github.com/kliment/Printrun/blob/master/buildinstructions.txt

probonopd, 2018-03-29:
> @probonopd, please correct me if I am wrong. Looking at the '.yml' files for programs such as Geany [0] or GIMP [1], it seems like if a 'printrun-x.x.deb' file exists in advance the 'Printrun.yml' could be as simple as

Yes... __but__ be sure to use "ingredient" debs built for Ubuntu 14.04, the equivalent Debian version, or earlier. Otherwise, the resulting AppImage will not run on all still-supported versions.

> I agree with @kliment, Printrun's AppImage should be built installing dependencies via pip in a manner similar to the one used for OSX [3].

Sure, that's also a great option. Can you install it to a venv? Then just put that venv into the AppImage. Remember, an AppImage is nothing more but a self-mounting filesystem image that executes whatever `AppRun` file is inside it.

Please have a look at https://github.com/AppImage/AppImageKit/wiki/Bundling-Python-apps for more information.

eduncan911, 2020-08-03:
Ping on this.  I almost exclusively run appimages or docker images.  Having issues "installing from source" on my clean Ubuntu 20.04 machine.

Yes, using virtualenv.
Yes, I pulled down `https://extras.wxpython.org/wxPython4/extras/linux/gtk3/ubuntu-20.04/wxPython-4.1.0-cp38-cp38-linux_x86_64.whl` and installed it.

However, it errors on `dbus-python` during requirements install.

kk6mrp, 2022-08-07:
I would also like to see this!