kliment/Printrun #813: Pronsole spams "wait" "ok 0"

jayproulx, 2017-04-25:
Hi everyone, really new here.  I'm running Pronsole from a Raspberry Pi (this will be my primary interface).  After connecting, there's quite a bit of "wait" "ok 0" spam on the console.  Is there a way to turn this off?  I quickly grepped through the code, and I'm not exactly sure where this is being printed from.

I'm not sure how helpful this is, maybe there's a really good reason for it, but I'm not sure what it is.  Seems like this just indicates that everything is A-OK and can be ignored.

Happy to send a pull request with options to mute it if someone can tell me what's going on here, it makes browsing through help, typing commands and reading other output rather annoying.

```offline> connect
No port specified - connecting to /dev/ttyUSB0 at 115200bps
start          
Printer is now online
Info:External Reset
Info: 0.000000 0.000000 0.000000 0.000000 0.000000 0.000000 0.000000 0.000000 0.000000
Free RAM:12337 
SD init fail   
wait           
ok 0           
wait           
wait           
ok 0           
wait           
wait           
wait           
ok 0           
wait           
wait           
ok 0           
wait           
wait           
ok 0           
wait           
```

jayproulx, 2017-04-25:
PR created and ready for review

iasj, 2017-05-16:
The last version of printrun is having this disabled. I was talking to Kliment on IRC, and he told me it was an option in the firmware shipped on you arduino. He made a quick fixed and now it doesn't keep printing anymore. Just clone the last version on master branch. 

I also installed the AUR/printrun-git in my Arch Linux, and it is working just fine. Check it out!

rockstorm101, 2017-10-21:
@jayproulx hi, this was supposed to be fixed by PR #814 and commit 9fa73a5583a9f2f1621163f1d08d3c849a3e4a98, could you please close this issue if it is really fixed? Thanks.

closed 2017-10-21 by jayproulx