kliment/Printrun #820: Pronterface registering as reader for *all* media types

gappleto97, 2017-05-05:
I'm pretty sure this program can handle neither python wheels nor blank cds. So why is it making file associations with these?

rockstorm101, 2017-05-05:
1. Probably a duplicate of #796.

 2. Could you give us some more info? What version of Printrun you are using, your OS, ...

gappleto97, 2017-05-06:
Kubuntu 17.04, using printrun 0~20150310 (which is the one in the Ubuntu repos). I don't know what MIME types it requested, but it ended up being registered for literally all of them.

rockstorm101, 2017-05-06:
Uhm, that's weird. I don't see any reasonable cause for this. Could you please give us the exact output of `dpkg -s printrun | grep -i version` and `cat /usr/share/applications/pronterface.desktop | grep -i mime`?

gappleto97, 2017-05-07:
Version: 0~20150310-3

MimeType=text/plain;application/octet-stream;application/sla;

gappleto97, 2017-05-07:
My guess would be it's the "application/octet-stream" causing the problem, and if so, it's definitely a duplicate of #796. Is there an easy way to test it on my end?

rockstorm101, 2017-05-07:
The version of printrun on Ubuntu 17.04 Zesty Zapus is actually 0~20150310-5ubuntu1 [[1]]. It seems you have and old version of the printrun package. Among other stuff, revision 5ubuntu1 includes a temporal fix for #796.

Your guess is right. If you were unable to upgrade printrun, a workaround would be to remove both "text/plain" and "application/octet-stream" mime types from '/usr/share/applications/pronterface.desktop':
```
- MimeType=text/plain;application/octet-stream;application/sla;
+ MimeType=application/sla;
```

[1]: https://launchpad.net/ubuntu/+source/printrun

closed 2017-05-07 by gappleto97