kliment/Printrun #822: Where are the userconf. in Linux

KC-Luke, 2017-05-14:
Hello,
i have change to 3d-view in the properties and now pronterface will not start. 

When i execute in console came this:

WARNING:root:Memory-efficient GCoder implementation unavailable: No module named gcoder_line
WARNING:root:Could not setup DBus for sleep inhibition: org.freedesktop.DBus.Error.ServiceUnknown: The name org.freedesktop.ScreenSaver was not provided by any .service files
WARNING:root:psutil unavailable, could not import power utils:No module named psutil

(python:9700): GLib-GObject-WARNING **: Attempt to add property GtkSettings::gtk-button-images after class was initialised
WARNING:root:RPC server bound on non-default port 7983

(python:9700): GLib-GObject-WARNING **: Attempt to add property GtkSettings::gtk-scrolled-window-placement after class was initialised

(python:9700): GLib-GObject-WARNING **: Attempt to add property GtkSettings::gtk-menu-bar-popup-delay after class was initialised

(python:9700): GLib-GObject-WARNING **: Attempt to add property GtkSettings::gtk-can-change-accels after class was initialised

(python:9700): GLib-GObject-WARNING **: Attempt to add property GtkSettings::gtk-menu-popup-delay after class was initialised

(python:9700): GLib-GObject-WARNING **: Attempt to add property GtkSettings::gtk-menu-popdown-delay after class was initialised

(python:9700): GLib-GObject-WARNING **: Attempt to add property GtkSettings::gtk-label-select-on-focus after class was initialised

(python:9700): GLib-GObject-WARNING **: Attempt to add property GtkSettings::gtk-menu-images after class was initialised

(python:9700): GLib-GObject-WARNING **: Attempt to add property GtkSettings::gtk-entry-select-on-focus after class was initialised

(python:9700): GLib-GObject-WARNING **: Attempt to add property GtkSettings::gtk-entry-password-hint-timeout after class was initialised
[ERROR] 3D view mode requested, but we failed to initialize it.
Falling back to 2D view, and here is the backtrace:
Traceback (most recent call last):
  File "/home/pi/Printrun/printrun/gui/viz.py", line 70, in __init__
    import printrun.gcview
  File "/home/pi/Printrun/printrun/gcview.py", line 22, in <module>
    from .gl.panel import wxGLPanel
  File "/home/pi/Printrun/printrun/gl/panel.py", line 27, in <module>
    import pyglet
ImportError: No module named pyglet

[ERROR] 3D view mode requested, but we failed to initialize it.
Falling back to 2D view, and here is the backtrace:
Traceback (most recent call last):
  File "/home/pi/Printrun/printrun/gui/viz.py", line 91, in __init__
    import printrun.gcview
  File "/home/pi/Printrun/printrun/gcview.py", line 22, in <module>
    from .gl.panel import wxGLPanel
  File "/home/pi/Printrun/printrun/gl/panel.py", line 27, in <module>
    import pyglet
ImportError: No module named pyglet

[WARNING] Could not load 3D viewer for plater:
Traceback (most recent call last):
  File "/home/pi/Printrun/printrun/stlplater.py", line 42, in <module>
    from printrun import stlview
  File "/home/pi/Printrun/printrun/stlview.py", line 22, in <module>
    import pyglet
ImportError: No module named pyglet

Traceback (most recent call last):
  File "/home/pi/Printrun/pronterface.py", line 35, in <module>
    app = PronterApp(False)
  File "/home/pi/Printrun/printrun/pronterface.py", line 2190, in __init__
    self.mainwindow = PronterWindow(self)
  File "/home/pi/Printrun/printrun/pronterface.py", line 218, in __init__
    self.reload_ui()
  File "/home/pi/Printrun/printrun/pronterface.py", line 277, in reload_ui
    self.createTabbedGui()
  File "/home/pi/Printrun/printrun/gui/__init__.py", line 184, in createTabbedGui
    from printrun.gcodeplater import GcodePlaterPanel
  File "/home/pi/Printrun/printrun/gcodeplater.py", line 34, in <module>
    from printrun.gl.libtatlin import actors
  File "/home/pi/Printrun/printrun/gl/libtatlin/actors.py", line 28, in <module>
    from pyglet.gl import glPushMatrix, glPopMatrix, glTranslatef, \
ImportError: No module named pyglet.gl

How i came back to 2d-view ?

kliment, 2017-05-14:
Can you try installing python-pyglet? With that installed, it will no
longer crash on startup, you'll have your 3d view, and will be able to
switch to 2d if you like.

On 14.05.2017 17:05, KC-Luke wrote:
> Hello,
> i have change to 3d-view in the properties and now pronterface will not start.
>
> When i execute in console came this:
>
> WARNING:root:Memory-efficient GCoder implementation unavailable: No [...]

KC-Luke, 2017-05-14:
No, I can´t install this, it doesn´t work

rockstorm101, 2017-12-14:
Hi, answering to the question in the title, the configuration file resides in `$HOME/.pronsolerc` on Linux systems.

Were you able to solve your problem? If so, could you share the solution with the community and close this issue?

rockstorm101, 2018-02-24:
I'm closing this issue. The question was answered and the OP did not complain. Please feel free to reopen if needed.

closed 2018-02-24 by rockstorm101