kliment/Printrun #824: TypeError: in method 'DC_SetPen', expected argument 2 of type 'wxPen const &'

hurzl, 2017-05-20:
Why do I get many of
```
Traceback (most recent call last):
  File "printrun/gviz.py", line 404, in paint
    self.repaint_everything()
  File "printrun/gviz.py", line 365, in repaint_everything
    self._drawarcs(dc, self.arcs[self.layerindex], self.arcpens[self.layerindex])
  File "printrun/gviz.py", line 322, in _drawarcs
    dc.SetPen(pens[i] if type(pens) == list else pens)
  File "/usr/local/lib/python2.7/site-packages/wx-2.8-gtk2-unicode/wx/_gdi.py", line 3662, in SetPen
    return _gdi_.DC_SetPen(*args, **kwargs)
TypeError: in method 'DC_SetPen', expected argument 2 of type 'wxPen const &'
```
?
Something is wrong with the list test as it works when I remove the ```if type(pens) == list else pens``` condition.

hurzl, 2017-05-20:
No, it only works with 
```pens[i] if hasattr(pens, "__getitem__") else pens```

rockstorm101, 2017-12-12:
Hi, could you please:

 * Check wether this also happens with the latest release?
 * State what OS and vesrion you are using? What version of wxPython?
 * Explain how to reproduce the error?
 
Thanks

closed 2018-03-02 by rockstorm101