kliment/Printrun #826: Latest release / Updated binaries

openhardwarecoza, 2017-05-27:
http://kliment.kapsi.fi/printrun/ still has builds from 2015 as "latest" but this repo has a lot of activity since

What is the reason new binaries arent being posted.  I know I can run from source / compile - but its just so much easier to grab a precompiled one?

apphead, 2017-06-05:
Couldn't agree more. Setting up the whole build environment feels like a waste of time if one does not change the code / contribute anything. What is stopping you from publishing a recent build for all platforms? Alpha beta gamma delta - does not matter ;-)

rockstorm101, 2017-06-24:
If help is needed to release a new version I'm willing to help. Just let me know what needs to be done. However I lack the resources to compile binaries for Windows and Mac OS X, if this is what's needed.

openhardwarecoza, 2017-06-24:
@alexrj @kliment

openhardwarecoza, 2017-06-24:
@xaon

openhardwarecoza, 2017-06-24:
@SparkyCola

apphead, 2017-06-24:
I'd like to run it on Windows 8 x64 or x86

SparkyCola, 2017-06-24:
I guess for those people that don't use the source a binary could be useful. However most advanced users would usually clone from git and run the install script (same for GNU/Linux-users?) and therefore always have the newest version anyway, plus it's python so you don't really (have to) compile there anyway.
However for less experienced users (and most Windows-users) this would be great.

openhardwarecoza, 2017-06-25:
[image]

So, i guess the http://kliment.kapsi.fi/printrun/ shttp://kliment.kapsi.fi/printrun/ is only maintened by @kliment himself? Any idea if he's still alive?

ntoff, 2017-07-08:
>However I lack the resources to compile binaries for Windows and Mac OS X

Those are probably the only 2 operating systems that precompiled versions would be the easier choice on since windows definitely doesn't come with python installed by default and the last time I tried to mess with python on OSX it required a LOT of messing around.

Not sure about OSX but I have it compiling under windows 7 (can't exactly test it fully since I'm running that via remote desktop so no opengl support for the 3d viewer), and it seems to run ok under windows 8 / 10.

The trouble is there seems to be a lot of changes required from the source available. 

Leaving the icon types as PNG works for running it from python source but compiling it results in an error, changing the source to use the .ico files fixes that. I'm unsure if that's a change that is absolutely required, or there's some other requirement not listed in requirements.txt that would make png icons work. 

The source also still lists the default slicer as skeinforge rather than slic3r as you find in the precompiled binaries. I don't know if the precompiled sources differ from those on github, whether there are specific changes necessary to make it compile on windows vs osx.

openhardwarecoza, 2017-07-08:
@kliment hey dude i see you did a commit 32 mins ago. At least have the decency to reply here about why not? Its your server after all?

openhardwarecoza, 2017-07-08:
[image]

rockstorm101, 2017-12-12:
I'm closing this issue since a new release has been published.

closed 2017-12-12 by rockstorm101