kliment/Printrun #834: setup.py does not install plugins

diggit, 2017-07-08:
Calling setup.py does not install plugins (whole directory missing) to their place. As result, pronterface.py does not start.

When running from cloned repository, pronterface.py starts normally.

```
Traceback (most recent call last):
  File "/usr/bin/pronterface.py", line 34, in <module>
    from printrun.pronterface import PronterApp
  File "/usr/lib/python2.7/site-packages/printrun/pronterface.py", line 32, in <module>
    from . import pronsole
  File "/usr/lib/python2.7/site-packages/printrun/pronsole.py", line 35, in <module>
    from . import printcore
  File "/usr/lib/python2.7/site-packages/printrun/printcore.py", line 41, in <module>
    from plugins import PRINTCORE_HANDLER
ImportError: No module named plugins
```

kliment, 2017-07-08:
@nexus511 can you do something about this?

nexus511, 2017-07-08:
Looks you got a point there. Thank you for hinting me. I will take a look at it.

nexus511, 2017-07-08:
Okay. To properly fix this, I will move the plugins directory into the printrun folder to ensure that it does not conflict with other site-packages.

I will implement this and provide a pull-request within the day. All in all it's not a big deal but might anyone using the plugin mechanism enforce to move her plugins once after updating. As I would guess that this is only me for now, I hope that this does not cause a lot of trouble.

kliment, 2017-07-08:
Fixed, thanks @nexus511

closed 2017-07-08 by kliment

diggit, 2017-07-08:
Thanks for fast solution!