kliment/Printrun #839: Pausing print on certain gcode command

chaneylc, 2017-08-04:
Using your printcore python module we are able to access the 'most recent(?)' gcode command with 

line = p.mainqueue.lines[p.queueindex].raw

Unfortunately this doesn't seem to be the actual currently printed command, is there a better way to access this? For example we would like to do something like this:

while p.printing:
    line = p.mainqueue.lines[p.queueindex].raw
    if "; arbitrary comment" in line:
        p.pause()
        do_work()
        p.resume()

Any thoughts or recommendations would be helpful.

Thanks

kliment, 2017-08-04:
you should use the sentcb callback instead for this - it gets triggered
whenever a command is sent

On 04.08.2017 02:15, Chaney Courtney wrote:
> Using your printcore python module we are able to access the 'most recent(?)' gcode command with
>
> line = p.mainqueue.lines[p.queueindex].raw
>
> Unfortunately this doesn't seem to be the actual [...]

rockstorm101, 2017-12-12:
Hi @chaneylc, if kliment's answer did solve your question and you consider the issue terminated, could you please close this issue?

Thanks

rockstorm101, 2018-02-24:
I'm closing this issue. The question was answered and the OP did not complain. Please feel free to reopen if needed.

closed 2018-02-24 by rockstorm101