kliment/Printrun #842: Printcore ERROR:root:Not connected to printer.

jarredglickstein, 2017-08-28:
This issue may be similar to the issue in #655 . I would like to write a simple python script to send short lists of commands to a printer. I have verified my computer is set up properly to communicate with the printer - I can use pronsole without any errors to connect to my printer and send GCode over the pronsole command line. 

Here is the test script I am using, taken directly from the current README.md:
```
from printrun.printcore import printcore
from printrun import gcoder
p=printcore('/dev/ttyACM0',115200) # or p.printcore('COM3',115200) on Windows
gcode=[i.strip() for i in open('filename.gcode')] # or pass in your own array of gcode lines instead of reading from a file
gcode = gcoder.LightGCode(gcode)
p.startprint(gcode) # this will start a print

#If you need to interact with the printer:
p.send_now("M105") # this will send M105 immediately, ahead of the rest of the print
p.pause() # use these to pause/resume the current print
p.resume()
p.disconnect() # this is how you disconnect from the printer once you are done. This will also stop running prints.
```
Here is the contents of filename.gcode (I'm not feeling terribly creative today):
```
G28 X
G90
G1 X 100
```
Once again for clarity, the expected behavior is as follows, and is observed if I use pronsole.py and manually enter the three lines: Printer homes X axis to 0, then moves X axis 100mm. 

Another potentially meaningful tidbit - here is the output of "python printcore.py /dev/ttyACM0 filename.gcode":

`Printing: filename.gcode on /dev/ttyACM0 with baudrate 115200`
No error message. The printer does not do anything.

jarredglickstein, 2017-08-28:
I made an accidental discovery. If I run pronsole, enter "connect /dev/ttyACM0 115200" then "exit", then run "python printcore.py /dev/ttyACM0 filename.gcode", the desired output is achieved - printcore will send the three lines of gcode from "filename.gcode" to the printer. If I attempt to run "python printcore.py /dev/ttyACM0 filename.gcode" a second time, nothing happens (and no lights blink on the arduino to indicate any data being transmitted or received). 

I can repeat a successful run of "python printcore.py /dev/ttyACM0 filename.gcode" by first opening pronsole and following the steps I listed above before each run of printcore.py. 

I tried this with the simple test script in my first post and again received the same error, "ERROR:root:Not connected to printer."

ajinkyapuar, 2017-10-01:
@jarredglickstein I've faced the same issue. 
@kliment https://stackoverflow.com/questions/46516954/send-one-line-of-g-code-to-a-arduino-based-3d-printer-using-a-python-script-over

parthaero, 2018-03-09:
Hi I am having the excatly the same issue. No Luck. Tried running the same code on two different computers.
Have you found a solution yet?

parthaero, 2018-03-09:
Also the stackoverflow link is no longer active. So if you guys did find a solution please let me know,
Thanks

kliment, 2018-03-09:
Hey, this sounds like an old driver issue I've seen before - can you try disconnecting, reconnecting again with another baudrate, disconnecting, reconnecting again with correct baudrate, and then sending? If that works then it's an issue with the ACM driver not picking up the baudrate correctly on first attempt, and might work with a newer pyserial version.

parthaero, 2018-03-09:
I tried what you said, didn't work. The thing is pronsole.py connects properly and I am able to operate the printer. I have checked for pyserial to be on latest version. Also I have tried running on a raspberry pi and on my desktop ubuntu, same issue. 
What should i do?

kliment, 2018-03-09:
Okay, can you try adding:
`while not p.online: time.sleep(0.1)`

before you send any commands, after you connect?

parthaero, 2018-03-10:
Tried this:

from printrun.printcore import printcore
import time

p = printcore()
p.connect('/dev/ttyUSB0', 250000)
while not p.online: time.sleep(0.1)
p.send_now("G28")
p.disconnect()

Same error:
WARNING:root:Memory-efficient GCoder implementation unavailable: No module named gcoder_line
ERROR:root:Not connected to printer

parthaero, 2018-03-10:
I tried this too:
python printcore.py /dev/ttyUSB0 filename.gcode
Does nothing. But Pronsole works. I'm guessing there is something pronsole is doing that printcore dosen't. I can't understand what.
Also I tried:
 ser = serial.Serial('/dev/ttyUSB0', 250000)  # open serial port
which also is working. I dont know how excatly to transfer gcode commands to test, but i know cause if i change the port to '/dev/ttyUSB1'(which is a wrong port) it gives an error 'ser' is not defined.
while '/dev/ttyUSB0' port executed the program without any errors.

parthaero, 2018-03-14:
The error "ERROR:root:Not connected to printer" vanishes when remove the command 'p.sendnow("G28")'.
running the following:
from printrun.printcore import printcore
p = printcore('/dev/ttyUSB3')
p.disconnect

Output:
WARNING:root:Memory-efficient GCoder implementation unavailable: No module named gcoder_line

kliment, 2018-03-14:
Well I tried it on a RAMPS machine running marlin and with the following code and it works fine (sends command to machine) so I don't know what's happening on your setup:

``
from printrun.printcore import printcore
import time
p = printcore()
p.connect('/dev/ttyACM3', 250000)
while not p.online: time.sleep(0.1)
p.send_now("G28")
p.disconnect()

parthaero, 2018-03-15:
I'm using a MKS GEN 1.4 in this machine.
Also pronsole.py works just fine.

parthaero, 2018-03-15:
running Marlin

VanessaE, 2018-03-16:
@kliment's code from his latest post works fine for me, also.

parthaero, 2018-03-18:
I’ll try again on another machine.

parthaero, 2018-03-24:
tried on a different machine, it uses the same MKS Gen 1.4 controller, but still the same issue. 
I made a new version of printcore:
from printrun.printcore import printcore
import os
import serial
import logging
import threading
from serial import *
from serial import SerialException
from functools import wraps

def locked(f):
    @wraps(f)
    def inner(*args, **kw):
        with inner.lock:
            return f(*args, **kw)

    inner.lock = threading.Lock()
    return inner

def control_ttyhup(port, disable_hup):
    """Controls the HUPCL"""
    if platform.system() == "Linux":
        if disable_hup:
            os.system("stty -F %s -hup" % port)
        else:
            os.system("stty -F %s hup" % port)

def enable_hup(port):
    control_ttyhup(port, False)

def disable_hup(port):
    control_ttyhup(port, True)

class comm():
    def __init__(self, port, baud):
        self.port = port
        self.baud = baud
        self.printer = None
        self.online = None
        self.printing = None
        if port is not None and baud is not None:
            try:
                self.connect(port, baud)
            except SerialException as e:
                print(e)
    @locked
    def disconnect(self):
        if self.printer:
            try:
                self.printer.close()
            except SerialException as e:
                print(e)
        self.printer = None
        self.online = None

    @locked
    def connect(self, port, baud):
        if self.printer:
            self.disconnect()
        if self.port is not None and self.baud is not None:
            try:
                self.printer = serial.Serial(self.port, self.baud)
#                print "Connecting to {} at {}", format(self.port, self.baud)
                self.printer.close()
            except SerialException as e:
                print(e)
            try:
                self.printer.open()
            except SerialException as e:
                print(e)
        self.read_thread = threading.Thread(target = self._listen)
        self.read_thread.start()

    def _listen_can_continue(self):
        return (self.printer
                and self.printer.isOpen())

    def _listen_until_online(self):
        while not self.online and self._listen_can_continue():
            self._send("M105")

    def _listen(self):
        if not self.printing:
            self._listen_until_online()

    def _checksum(self, command):
        return reduce(lambda x, y: x ^ y, map(ord, command))

    def _send(self,command,lineno = 0, calcchecksum = False):
        if calcchecksum:
            prefix = "N" + str(lineno) + " " + command
            command = prefix + "*" + str(self._checksum(prefix))
            if "M110" not in command:
                self.sentlines[lineno] = command
`
This worked. Everything else i've kept same except the 'connect' function.
Has it got something to do with pySerial?

kliment, 2018-03-24:
Sorry, the formatting is a bit strange there. Can you post just your connect function?

parthaero, 2018-03-24:
@locked
def connect(self, port, baud):
    if self.printer:
        self.disconnect()
    if self.port is not None and self.baud is not None:
        try:
            self.printer = serial.Serial(self.port, self.baud)
print "Connecting to {} at {}", format(self.port, self.baud)
            self.printer.close()
        except SerialException as e:
            print(e)
        try:
            self.printer.open()
        except SerialException as e:
            print(e)
    self.read_thread = threading.Thread(target=self._listen)
    self.read_thread.start()

sorry about the formatting i dont know whats going on there.

kliment, 2018-03-24:
So it looks like you need to open the port twice to make it work. I don't know why that's what it takes on your setup but it works with the unmodified file for everyone else so I expect it's either a serial driver or pyserial issue. But your modified printcore file solves your problem, right?

parthaero, 2018-03-25:
If you notice this line:
self.printer = serial.Serial(self.port, self.baud)
I am importing serial directly for using it here.

            else:
                disable_hup(self.port)
                self.printer_tcp = None
                try:
                    self.printer = Serial(port = self.port,
                                          baudrate = self.baud,
                                          timeout = 0.25,
                                          parity = PARITY_ODD)
                    self.printer.close()
                    self.printer.parity = PARITY_NONE
                    try:  #this appears not to work on many platforms, so we're going to call it but not care if it fails
                        self.printer.setDTR(dtr);
                    except:
                        #self.logError(_("Could not set DTR on this platform")) #not sure whether to output an error message
                        pass
                    self.printer.open()
                except SerialException as e:
                    self.logError(_("Could not connect to %s at baudrate %s:") % (self.port, self.baud) +
                                  "\n" + _("Serial error: %s") % e)
                    self.printer = None
                    return
                except IOError as e:
                    self.logError(_("Could not connect to %s at baudrate %s:") % (self.port, self.baud) +
                                  "\n" + _("IO error: %s") % e)
                    self.printer = None
                    return
            for handler in self.event_handler:
                try: handler.on_connect()
                except: logging.error(traceback.format_exc())
            self.stop_read_thread = False
            self.read_thread = threading.Thread(target = self._listen)
            self.read_thread.start()
            self._start_sender()

I think even your function opens the port twice. after using the:
 self.printer = Serial(port = self.port,
                                          baudrate = self.baud,
                                          timeout = 0.25,
                                          parity = PARITY_ODD)
it again closes the port, opening it again after setDTR.

And i tried making the change in your files it didn't work, I rewrote the file.
Correct me if I am wrong.

kliment, 2018-03-25:
You're not wrong but I don't really see what you're doing differently. Anyway, it works for me and for almost everyone else with the original code, and it works for you with your code, so I think we can consider this issue resolved.

closed 2018-03-25 by kliment

amrithmmh, 2019-02-15:
> tried on a different machine, it uses the same MKS Gen 1.4 controller, but still the same issue.
> I made a new version of printcore:
> `import os
> import serial
> import logging
> import threading
> from serial import *
> from serial import SerialException
> from functools import wraps
> 
> def locked(f):
> @wraps(f)
> def inner(*args, **kw):
> with inner.lock:
> return f(*args, **kw)
> 
> ```
> inner.lock = threading.Lock()
> return inner
> ```
> def control_ttyhup(port, disable_hup):
> """Controls the HUPCL"""
> if platform.system() == "Linux":
> if disable_hup:
> os.system("stty -F %s -hup" % port)
> else:
> os.system("stty -F %s hup" % port)
> 
> def enable_hup(port):
> control_ttyhup(port, False)
> 
> def disable_hup(port):
> control_ttyhup(port, True)
> 
> class comm():
> def **init**(self, port, baud):
> self.port = port
> self.baud = baud
> self.printer = None
> self.online = None
> self.printing = None
> if port is not None and baud is not None:
> try:
> self.connect(port, baud)
> except SerialException as e:
> print(e)
> @locked
> def disconnect(self):
> if self.printer:
> try:
> self.printer.close()
> except SerialException as e:
> print(e)
> self.printer = None
> self.online = None
> 
> ```
> @locked
> def connect(self, port, baud):
>     if self.printer:
>         self.disconnect()
>     if self.port is not None and self.baud is not None:
>         try:
>             self.printer = serial.Serial(self.port, self.baud)
> ```
> # print "Connecting to {} at {}", format(self.port, self.baud)
> ```
>             self.printer.close()
>         except SerialException as e:
>             print(e)
>         try:
>             self.printer.open()
>         except SerialException as e:
>             print(e)
>     self.read_thread = threading.Thread(target=self._listen)
>     self.read_thread.start()
> 
> def _listen_can_continue(self):
>     return (self.printer
>             and self.printer.isOpen())
> 
> 
> def _listen_until_online(self):
>     while not self.online and self._listen_can_continue():
>         self._send("M105")
> 
> 
> def _listen(self):
>     if not self.printing:
>         self._listen_until_online()
> 
> 
> def _checksum(self, command):
>     return reduce(lambda x, y: x ^ y, map(ord, command))
> 
> def _send(self,command,lineno = 0, calcchecksum = False):
>     if calcchecksum:
>         prefix = "N" + str(lineno) + " " + command
>         command = prefix + "*" + str(self._checksum(prefix))
>         if "M110" not in command:
>             self.sentlines[lineno] = command
> ```
> `
> 
> This worked. Everything else i've kept same except the 'connect' function.
> Has it got something to do with pySerial?

can you share the printcore file you have i cannot get any output only send commands

lunknowl, 2021-06-23:
can you share the file?

> You're not wrong but I don't really see what you're doing differently. Anyway, it works for me and for almost everyone else with the original code, and it works for you with your code, so I think we can consider this issue resolved.

lunknowl, 2021-06-23:
> If you notice this line:
> self.printer = serial.Serial(self.port, self.baud)
> I am importing serial directly for using it here.
>
> ```
>         else:
>             disable_hup(self.port) [...]

can you share the file?