kliment/Printrun #843: Mistake

Digggrrr, 2017-08-30:
Hi, Everyone. First time and I thought I'd try this for a Topic: When Closing Pronterface, is it Possible to add code for a "ARE YOU SURE"??? button. You know ;) so Someone Doesn't close down the program, say Mid-Print...

rockstorm101, 2017-12-12:
This request is similar to #825 and might have a common solution. I'm renaming it so the title is more self explanatory.

kliment, 2018-03-24:
@Digggrrr can you check what happens when you try to close pronterface mid-print using file->exit rather than closing the window?

sledjunk, 2020-07-12:
I tried this today in RC5.  When I try to close the app, either by exiting or via the 'X', the print pauses, but does not change the pause button to resume, sets the nozzle temp to 0 and the printer continues to go through the motions of printing, eventually reporting, "Cold extrusion prevented".  This does not stop the printer from continuing to go through its motions.
This is in Win7 with a Prusa I3 MkIII.
Hope this helps.