kliment/Printrun #848: Repetier firmware configured with SD_EXTENDED_DIR=1 M20 returns filenames with sizes, sdprint <filename> fails

paulfertser, 2017-09-20:
Example M20 result line:
```
...
digi-console-holder.gcode 89798
...
```
This makes sdprint require not only filename but also size because this is what gets stored in the sd card file list.

I suggest to strip off a decimal number if it's appended after a space to a filename.

kliment, 2018-03-23:
This should be fixed in the 2.x branch - can you verify?

paulfertser, 2018-03-24:
```
Python 3.5.4 (default, Feb  4 2018, 13:48:39) 
[GCC 6.4.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import re
>>> line = "digi-console-holder.gcode 236775"
>>> re.sub(" \d+","",line.strip().lower())
'digi-console-holder.gcode'
>>> line = "digi-console-holder - 2.gcode 236775"
>>> re.sub(" \d+","",line.strip().lower())
'digi-console-holder -.gcode'
>>> re.sub(" \d+$","",line.strip().lower())
'digi-console-holder - 2.gcode'
>>> 
```

kliment, 2018-03-24:
Well spotted, just changed that.

paulfertser, 2018-03-24:
Thank you. My specific testcase was with a uSD card that came with Anet A8 and Repetier FW that I installed to it. I hoped to use pronsole to script enabling the bed heater, then uploading a file to the card, then starting the actual print (so that relatively slow upload speed wouldn't matter as the bed takes time to heat no matter what). However, the firmware reports busy when heating the bed and so pronsole doesn't try uploading during that, heh. Regarding the filenames, I haven't removed anything from the card and here's what the firmware replies to M20:
```
ok 0                                                        
Begin file list                                             
-ԇD/                                                        
-D/Å(ô/                                                    
-ŇD/Å(ô/A8Í\ô-ŇH       -1.1.pdf 9180456                    
-D/Å(ô/A8cöþŇÅ-1.2.pdf 6161156                             
-ÅD/Å(ô/-Ň3DSp(ÏîϘ ã³W-anet1.0.pd 3397932
-㳇D/Sp!STL/
-D/Sp!STL/ºi.stl 70390780
-D/Sp!STL/'}.stl 29876784
-D/Sp!STL/ÒP.stl 7889
-҇D/Sp!STL/.STL 335184
-D/Sp!STL/R.stl 21810584
-D/Sp!STL/aË.stl 197184
-ˇD/Sp!STL/ÑWT.stl 1973784
-чD/KÕՇöGCODE/
-D/KÕ?öGCODE/Baymax-1.75mm PLA-A8.gcode 10269757
-D/KÕ?öGCODE/Box-1.75mm PLA-A8.gcode 236023
-D/KÕ?öGCODE/brush barrel-1.75PLA-A8.gc 27757004
-D/KÕ?GCODE/chess-1.75mm PLA-A8.gcode 1519165
-D/KÕ?öGCODE/Figure-1.75mm PLA-A8.gcode 4391660
-D/KÕ?öGCODE/FU-1.75mm PLA-A8.gcode 1458217
-D/KÕ?öGCODE/Pyramid-1.75MM PLA-A8.gcod 5619530
-D/oö/
-?D/MöUÊvþG/
-D?/MöUÊvþG/3DSp:A8ÅMööùgh-1.2.pdf 759388
-D?/MöUÊvþG/A8 3DSp:ÅÆ?Q@.pdf 233855
-D/MUÊvþG/-;.jpg 3395889
-?D/MöUÊvþG/ 460738
English Instructions/
English Instructions/Installation Instruction/
English Instructions/Installation Instruction/A8 3D Printer Installation 7839313
English Instructions/Installation Instruction/A8 Operation instruction-1 10883866
English Instructions/Installation Instruction/Print Quality Troubleshoot 3369900
...
```
This obviously leads to "Got rubbish reply" from pronsole when it tries to parse this reply. Also, all filenames are truncated to 26 characters.
Just so you know...
Probably I should switch to Marlin? But I thought Repetier had a clearer codebase.

closed 2018-03-24 by paulfertser

kliment, 2018-03-24:
You can set the bed temperature without waiting for it to reach it, do
your upload, then send another command to wait for temp. Unfortunately
as we can't detect what encoding the filenames are in there is no easy
way around the "rubbish reply" so all I can advise you to do is to
remove non-ascii filenames.

paulfertser, 2018-03-24:
> You can set the bed temperature without waiting for it to reach it, do
your upload, then send another command to wait for temp
I know it should be possible, and I have seen it working few times, and few times not working (with the same script), and I was too demotivated to dig it further, I had an impression it's the firmware itself preventing the upload. BTW, the gcode already has command for waiting for the setpoint, so starting sd print should be enough.

kliment, 2018-03-24:
Well I can't help you with your firmware unfortunately, but thanks for helping make printrun better.

paulfertser, 2018-03-24:
You might want to consider this usecase as essential and add a special feature to printrun so that a user could just press a button and have his or her bed heated while transferring the gcode to SD card and then print it safely from there. I've seen few requests like that on "the forums". Then if some firmware would be misbehaving it would be easier to spot and fix, I guess.

kliment, 2018-03-24:
I don't see the use case as essential at all - most people that do that sort of thing make a macro to preheat everything and have a button trigger that macro, and everyone seems to have their own opinion of what should be happening when preparing for a print, and that's okay. I'd rather not add a special case like this.

paulfertser, 2018-03-24:
Yeah, ok, I'm not someone to judge, thanks for your attention :)

commit 2c6d3e65 (2018-03-23): Ignore file sizes in incoming SD filenames - fixes #848
commit 1bd01451 (2018-03-24): Fix #848 - only remove number at end of filename