kliment/Printrun #849: ls fails to report filelist if it contains non-utf8 characters

paulfertser, 2017-09-20:
Example data from Repetier firmware showing the listing of an Anet A8 factory SD card:
```
00000060  77 61 69 74 0d 0a 6f 6b  20 30 0d 0a 42 65 67 69  |wait..ok 0..Begi|
00000070  6e 20 66 69 6c 65 20 6c  69 73 74 0d 0a 2d 87 44  |n file list..-.D|
00000080  99 2f 0d 0a 2d 87 44 99  2f 89 c5 7f 28 f4 0e 2f  |./..-.D./...(../|
00000090  0d 0a 2d 87 44 99 2f 89  c5 7f 28 f4 0e 2f 41 38  |..-.D./...(../A8|
000000a0  cd 5c f4 0e 66 08 2d 87  48 09 2d 31 2e 31 2e 70  |.\..f.-.H.-1.1.p|
000000b0  64 66 20 39 31 38 30 34  35 36 0d 0a 2d 87 44 99  |df 9180456..-.D.|
000000c0  2f 89 c5 7f 28 f4 0e 2f  41 38 63 f6 fe 87 89 c5  |/...(../A8c.....|
000000d0  f4 0e 2d 31 2e 32 2e 70  64 66 20 36 31 36 31 31  |..-1.2.pdf 61611|
000000e0  35 36 0d 0a 2d 87 44 99  2f 89 c5 7f 28 f4 0e 2f  |56..-.D./...(../|
000000f0  2d 87 33 44 53 70 28 cf  ee 98 20 e3 b3 07 57 2d  |-.3DSp(... ...W-|
00000100  61 6e 65 74 31 2e 30 2e  70 64 20 33 33 39 37 39  |anet1.0.pd 33979|
00000110  33 32 0d 0a 2d 87 44 99  2f 53 70 21 8b 53 54 4c  |32..-.D./Sp!.STL|
00000120  2f 0d 0a 2d 87 44 99 2f  53 70 21 8b 53 54 4c 2f  |/..-.D./Sp!.STL/|
00000130  ba 69 2e 73 74 6c 20 37  30 33 39 30 37 38 30 0d  |.i.stl 70390780.|
```
This makes pronsole fail to report anything useful whatsoever:
```
ttyUSB0 PC> Files on SD card:
[ERROR] Traceback (most recent call last):
  File "/home/pavel/tmp/Printrun/printrun/printcore.py", line 268, in _readline
    try: self.recvcb(line)
  File "/home/pavel/tmp/Printrun/printrun/pronsole.py", line 1285, in recvcb
    listener(l)
  File "/home/pavel/tmp/Printrun/printrun/pronsole.py", line 1081, in listfiles
    self.log("\n".join(self.sdfiles))
  File "/home/pavel/tmp/Printrun/printrun/pronsole.py", line 257, in log
    msg = u"".join(unicode(i) for i in msg)
  File "/home/pavel/tmp/Printrun/printrun/pronsole.py", line 257, in <genexpr>
    msg = u"".join(unicode(i) for i in msg)
  File "/usr/lib/python2.7/encodings/utf_8.py", line 16, in decode
    return codecs.utf_8_decode(input, errors, True)
UnicodeDecodeError: 'utf8' codec can't decode byte 0x87 in position 1: invalid start byte

```
despite the fact some other properly named files are present later in the listing.

kliment, 2018-03-24:
Can you let me know what error this gives on the 2.x branch?

paulfertser, 2018-03-24:
``Got rubbish reply from /dev/ttyUSB0 at baudrate 500000:''

I suggest the following workaround: when processing the reply line-by-line just ignore those that can't be interpreted as utf-8 (I think restricting to ASCII here would be unreasonable). Then "normal" files would still be visible, "sdprint" work as expected etc. Alternatively, you might want to add some explicit words to make your users always start with a clean empty filesystem on the SD card? Otherwise it might get really confusing, especially for novices, I'm afraid.

kliment, 2018-03-24:
The issue is that doing so would break the bad baudrate detection - we can of course ignore non-ascii stuff coming in but since moving to py3 we now do the encoding/decoding much earlier in the chain, so at the time of decoding we don't know if the firmware is sending filenames or not. We don't expect firmware to be sending us anything non-ascii, so the options here are a) disable bad baudrate detection, remove non-ascii characters (resulting in weird or empty filenames sometimes) or b) keep it as it is now. I'm not sure which is the better option. I do know a lot more people attempt to connect at the wrong baudrate than there are total people who print from SD using printrun, so given this affects a fraction of the smaller group but the fix would affect a much larger group, I'd rather keep it as is. @hroncok, @rockstorm101 do you have an opinion on this?

hroncok, 2018-03-25:
Well, we can workaround it with words.

> Got rubbish reply from /dev/ttyUSB0 at baudrate 500000. Either the port/baudrate combination is wrong or you are trying to list filenames from SD card that are not properly encoded (start with an empty SD card to fix that). 

In case the text is too long and specific, it can be replaced with:

> Got rubbish reply from /dev/ttyUSB0 at baudrate 500000. Try using different baudrate. See list of other possible causes at <URL>.