kliment/Printrun #850: Shebang line contains just "python" but code can't be run with Python3

paulfertser, 2017-09-20:
On systems where python3 is the default Python interpreter the shebang line makes OS execute pronsole and other scripts using python3 which fails. Probably the shebang line should be changed to always use python2.

rockstorm101, 2017-12-12:
Hi, could it be as simple as replacing every `#!/usr/bin/env python` to `#!/usr/bin/python2`? It works on Debian systems, and looking at Fedora's python packaging policy [[1]] it should work too. (@hroncok, please correct me if I'm wrong)

@kliment, would this break Windows or Mac OS?

[1]: https://fedoraproject.org/wiki/Packaging:Python#Multiple_Python_Runtimes

kliment, 2017-12-12:
Not sure about whether osx has a /usr/bin/python2 but "/usr/bin/env
python2" will work everywhere that "/usr/bin/env python" currently does

On 12.12.2017 16:22, Rock Storm wrote:
> Hi, could it be as simple as replacing every `#!/usr/bin/env python` to `#!/usr/bin/python2`? It works on Debian systems, and looking at Fedora's python packaging policy [[1]] it should work too. [...]

rockstorm101, 2017-12-12:
It seems the form `/usr/bin/env <whatever>` is now deprecated in favor of `/usr/bin/pythonX`, at least on Debian systems [[1]] and Fedora systems [[2]].

[1]: https://www.debian.org/doc/packaging-manuals/python-policy/ch-programs.html
[2]: https://fedoraproject.org/wiki/Packaging:Python#Multiple_Python_Runtimes

kliment, 2017-12-12:
Okay, let's go with that then

On 12.12.2017 17:03, Rock Storm wrote:
> It seems the form `/usr/bin/env <whatever>` is now deprecated in favor of `/usr/bin/pythonX`, at least on Debian systems [[1]] and Fedora systems [[2]].
>
> [1]: [...]

hroncok, 2017-12-12:
Well it's deprecated on a packaging level, not for upstream projects.

IMHO Best thing here is to forgot all scripts and shebangs at all and use entry_points instead.

rockstorm101, 2017-12-13:
That'd be fine by me too. It's probably a more elegant solution.

hroncok, 2018-03-20:
Shebangs have python3 in the 2.x branch.

closed 2018-03-20 by hroncok