kliment/Printrun #856: UI issue with extrude settings.

baznz, 2017-10-09:
In the UI, extrude length and extrude speed settings are not lined up with the text "Length:" and "Speed:". Instead they appear to be a little higher where the Extrude and Reverse buttons paint over the top of them, rendering them invisible and impossible to read/set. I can just make out the outline of what appears to be these two mis-placed fields while I re-size the Printrun window by dragging on its bottom edge. 

Currently I work around this issue by editing printrunconf.ini (set last_extrusion), but you have to live with only one set extrude  length and speed. I learned this from #684.   
 
Printrun 2014.08.01 Precompile. Windows 8.1 64bit

rockstorm101, 2017-12-12:
Hi, I am currently unable to reproduce this, could you check whether this also happens with the 1.6.0 version?. And if so, could you provide a screenshot of it?

Thanks

baznz, 2017-12-27:
Hi rockstorm, I unpacked and run 1.6 from here:
`http://kliment.kapsi.fi/printrun/`
and still see the same issue.
[image]
I shot a short video of Printrun and re-sized the bottom edge of Printrun in the video. Screnshot1 is taken from this video showing my Printrun window with nothing moving.

[image]
Screenshot2 and 3 show some artifacts that occasionally appear while the Printrun window is dragged to re-size it. I included them in case the offers a further clue to what might be happening. 
[image]
I hope this helps.

Fonskee, 2017-12-27:
Hi,
I had the same issue with pronterface. The problem seemed to be my high resolution screen.

I solved it by right clicking  on the executable pronterface.exe, go to properties, compatibility tab and check "Override high DPI scaling behavior. Scaling performed by:" and select "application" in the dropdown menu.

Hope this helps!

rockstorm101, 2017-12-27:
Hi @Fonskee, thanks a lot for your feedback I really appreciate it. I'm still unable to reproduce the issue, could this workaround work for you @baznz?

baznz, 2018-01-04:
Thanks @Fonskee, you tip solves the issue for me. @rockstorm101 it works for both the Printrun versions listed here.
For others with Windows 8.1 the compatibility setting seems a little less detailed - just a single checkbox "Disable display scaling on high DPI settings". It probably makes sense for people with administrator rights to go "Change settings for all users" first, then apply the setting.

closed 2018-01-04 by baznz

seelipraneeth, 2019-08-03:
> Hi,
> I had the same issue with pronterface. The problem seemed to be my high resolution screen.
>
> I solved it by right clicking on the executable pronterface.exe, go to properties, compatibility tab [...]

It is really helpful, and working for me. thanks for saving a lot of time

Nidhjin, 2020-03-23:
> I solved it by right clicking on the executable pronterface.exe, go to properties, compatibility tab and check "Override high DPI scaling behavior. Scaling performed by:" and select "application" in the dropdown menu.

Thanks a lot for that ! Still helpfull in 2020 ;)

Regaor, 2020-04-24:
Thanks a lot, it worked  very well by activating as in the picture!

[image]