kliment/Printrun #857: AttributeError: 'Colour' object has no attribute 'SetFromName'

Sineos, 2017-10-20:
Trying to run from source on Windows 10 but getting following error:

```
D:\21_Git\Printrun\printrun\pronterface.py:728: wxPyDeprecationWarning: Call to deprecated item. Use Append instead.
  m.AppendMenu(wx.ID_ANY, _("&Recent Files"), recent)
Traceback (most recent call last):
  File "D:\21_Git\Printrun\pronterface.py", line 66, in <module>
    app = PronterApp(False)
  File "D:\21_Git\Printrun\printrun\pronterface.py", line 2296, in __init__
    self.mainwindow = PronterWindow(self)
  File "D:\21_Git\Printrun\printrun\pronterface.py", line 220, in __init__
    self.reload_ui()
  File "D:\21_Git\Printrun\printrun\pronterface.py", line 282, in reload_ui
    self.settings.controlsmode == "Mini")
  File "D:\21_Git\Printrun\printrun\gui\__init__.py", line 234, in createGui
    controls_sizer = ControlsSizer(self, controls_panel, mini_mode = mini)
  File "D:\21_Git\Printrun\printrun\gui\controls.py", line 387, in __init__
    else: self.make_standard(root, parentpanel, standalone_mode)
  File "D:\21_Git\Printrun\printrun\gui\controls.py", line 395, in make_standard
    self.xyzsizer = XYZControlsSizer(root, xyzpanel)
  File "D:\21_Git\Printrun\printrun\gui\controls.py", line 31, in __init__
    root.xyb = XYButtons(parentpanel, root.moveXY, root.homeButtonClicked, root.spacebarAction, root.bgcolor, zcallback=root.moveZ)
  File "D:\21_Git\Printrun\printrun\gui\xybuttons.py", line 71, in __init__
    self.bgcolor.SetFromName(bgcolor)
AttributeError: 'Colour' object has no attribute 'SetFromName'
```

Any pointer would be appreciated. Thanks

concifederico, 2017-11-21:
Same error on Ubuntu 17.04

concifederico, 2017-11-21:
The answer is to replace every
self.bgcolor.SetFromName(bgcolor)
to
self.bgcolor.Set(bgcolor)
Regards,

Wurstnase, 2017-11-23:
Printrun is developed in the past with wxPython 3x. With wxPython 4x there are a lot of changes. Code needs to be changed. So doing the change by yourself or installing the old wxPython 3.

kliment, 2018-03-08:
This is fixed in the 2.x branch, which builds against wxpython4

closed 2018-03-08 by kliment

spase84, 2018-10-07:
@kliment how can i find 2.x branch? I have same issue on ubuntu 18.04

rockstorm101, 2018-10-07:
@spase84, are you using the packaged printrun? Beacuse the one packaged in Ubuntu is still at version 1.6.0. To use version 2.x you must run Printrun [from source](https://github.com/kliment/Printrun#running-from-source).