kliment/Printrun #863: How to Compile Source into a Standalone Portable

kcheeeung, 2017-11-08:
Hi everyone. Are there recommended instructions as to ways I can package this into a standalone portable file example .exe files for Windows ect?

rockstorm101, 2017-12-12:
Hi, I'm not sure I understand what you ask, the pre-compiled binary provided for Windows ([Printrun-win-18Nov2017.zip][1]) works right out of the box. Just unzip it and run pronterface.exe, no installation needed.

Or are only the instructions of how to compile this binary what you need?

[1]: https://github.com/kliment/Printrun/releases/download/printrun-1.6.0/Printrun-win-18Nov2017.zip

kcheeeung, 2017-12-12:
Yes you're right. I just would like instructions on how to compile the binary into an exe file or mac app file.

It might be helpful for other people to add documentation on the instructions about compiling the source code into a portable file (exe or mac app file). Before the latest "pre-compiled release", I remember the last pre-compiled version was from 2015 and there has been many updates since then.

kliment, 2018-03-23:
The 2.x branch now contains instructions for building binaries on both windows and osx, as well as installing necessary dependencies.

kliment, 2018-03-23:
See https://github.com/kliment/Printrun/blob/2.x/buildinstructions.txt

closed 2018-03-23 by kliment