kliment/Printrun #864: Printrun-Mac-18Nov2017 does not open macOS High Sierra

Eternyt, 2017-11-21:
Simply download the precompiled binaries for Mac, unzipped and double-clicked. Then nothing happens and if I go to the force quit utility it says that printrun is not responding. Any suggestion?

Eternyt, 2017-11-26:
Probably. I confirm that I run on macOS High Sierra too.

kliment, 2017-11-26:
Looks like something changed fundamentally between Sierra and High Sierra. I have no idea how to investigate the issue, but apparently 32 bit applications have problems on HS.

loelkes, 2017-12-02:
I get this error message
 
> Computer:MacOS loelkes$ ./pronterface
Traceback (most recent call last):
  File "<string>", line 35, in <module>
  File "/Users/craigvanvliet/Desktop/PyInstaller-2.1/pronterface/build/pronterface/out00-PYZ.pyz/printrun.pronterface", line 2190, in __init__
  File "/Users/craigvanvliet/Desktop/PyInstaller-2.1/pronterface/build/pronterface/out00-PYZ.pyz/printrun.pronterface", line 219, in __init__
  File "/Users/craigvanvliet/Desktop/PyInstaller-2.1/pronterface/build/pronterface/out00-PYZ.pyz/printrun.pronterface", line 901, in update_recent_files
  File "/Users/craigvanvliet/Desktop/PyInstaller-2.1/pronterface/build/pronterface/out00-PYZ.pyz/wx._misc", line 916, in AddFileToHistory
wx._core.PyAssertionError: C++ assertion "Assert failure" failed at /BUILD/wxPython-src-2.9.4.0/src/common/stockitem.cpp(214) in wxGetStockLabel(): invalid stock item ID

>   System Version:	macOS 10.13.1 (17B48)
  Kernel Version:	Darwin 17.2.0

I took the zip from the website and startet it via terminal. Same error for 18Nov2017 and 03Feb2015

drifterdon, 2018-01-07:
Same thing here

cinealfa, 2018-02-10:
Same problem, I guess no solutions?

mlazarin, 2018-02-19:
Guys, had an issue with Pronterface here in my Mac (High Sierra). It was working, and then suddenly didn't open anymore. After a lot of Google search, I found (in a forum about using Pronterface in Ubuntu) the information about a .pronsolerc file in my home directory (/Users/<your username>), and suspected about corruption of the file. Removed the 2 copies there (.pronsolerc and .pronsolerc˜bak), and after that Pronterface came back alive again!! So if at any time it already opened and stopped to open anymore, you can try this. Even if it never worked, I think trying this will not hurt.
Good luck from Brazil!!!

mlazarin, 2018-02-19:
And yes, Pronterface works in MacOS High Sierra! :)

ololoken, 2018-02-23:
on my os x high sierra printrun launched after adding "localhost 127.0.0.1" to /etc/hosts

Fsarmento, 2018-03-17:
Thanks @mlazarin, problem solved

kliment, 2018-03-20:
Does this work for everyone? Can I consider this issue resolved?

Eternyt, 2018-03-20:
Localhost fixed my problem, now I'm able to open and use Printrun 1.6.0 (the last one stable). I also removed the two files as suggested by @mlazarin  (even if now they are present again) but not checked immediately so perhaps also this suggestion could have some benefit.
I will keep this issue open for another week to give the time to the other people with this problem to try the solutions suggested, and then if it works for all I will close it (if the owner have other ideas obviously say them).

kliment, 2018-03-26:
Can you please test with the latest OSX binary, released as 2.0.0rc5?

see3d, 2018-03-27:
The previous binary I ran was 1.6 — after deleting the two files posted by miazarin, it started up and sort of ran, but crashed so many times and had so many UI bugs that I had to uninstall it to keep from accidentally running it again.  

It took a while, but I found the binary for 2.0.0rc5. 
I was able to make it run, connect to my delta MPMD printer and execute most of the UI commands.  It seems to be working without any problems so far.

Thank you.  I will start using it for everyday interfacing now.  

I am intrigued by the macro functionality.  It looks like it could help me do some auto calibration, but I don’t see a way to get to the feedback data from a G29 V4 command.

Dennis

Eternyt, 2018-03-27:
@kliment seems to work too, but I didn't test all the feature. I have to highlight what @see3d experienced because happened the same UI glitch on my mac.

closed 2018-03-27 by Eternyt

kliment, 2018-03-28:
There is now a signed DMG file on the releases page - can someone please test whether they can install from that file on OSX?

michaelgotchosen, 2018-05-02:
same problem here. localhost fixed it.

GeminiServer, 2018-12-21:
I was having the same issue. Touched the /etc/hosts with the localhost but i was not able to ping localhost in the Terminal.

1. Ping
>>ping localhost
>> error ping: cannot resolve localhost: Unknown host

2. Change hosts file:
>> sudo nano /etc/hosts

3. Add localhost with 4 spaces to separate. Save it and restart the terminal.
 >> 127.0.0.1{4spaces}localhost 

4. To confirm if everything works fine, try pinging your localhost
>>ping localhost

Printrun will now start.