kliment/Printrun #865: object has no attribute 'slic3r_configs'

SteveDC2, 2017-12-11:
I am using the Windows binary distributions from http://kliment.kapsi.fi/printrun/

I have tried both 2015 and 2017 versions but both give me the following error when I try to load an STL file.

Thoughts?

  File "printrun\pronterface.pyc", line 1298, in slice_func
AttributeError: 'PronterWindow' object has no attribute 'slic3r_configs'

rockstorm101, 2017-12-12:
Hi, I am trying to reproduce this behaviour, could you state the version of windows you are using and paste here the exact line of _Settings_ > _Options_ > _External commands_ > _Slice command_ ?

Thanks

SteveDC2, 2017-12-13:
So, I can't seem to repeat the error now either!!

I was experimenting with various other builds and other slicers (Repetier, Cura etc...) and now Printrun seems to work fine.

I will try doing a clean install on another PC to see if I can 'recover' the failure.

By the way, I really appreciate you responding to my issue so quickly. 

BR,
Steve

helaan, 2018-01-27:
I've just hit this myself, repro steps:

1. Disable Slic3r integration in Settings -> User Interface -> [ ] Enable Slic3r integration
2. Restart Pronterface
3. Enable Slic3r integration in Settings -> User Interface -> [ ] Enable Slic3r integration
4. Load an STL file. 

(When starting pronterface with slic3r integration enabled, it works fine, which is why OP couldn't reproduce)

OS: Gentoo

kliment, 2018-03-23:
I believe this is now fixed in the 2.x branch (that requires python3 and wxpython 4). At least I can no longer reproduce there with the above repro steps.

SteveDC2, 2018-06-25:
Thanks for looking at this. I will check ASAP.

commit a280f44e (2018-03-23): Dynamically rebuild menu when options change - should fix #865