kliment/Printrun #867: The spool manager makes pronsole depend on wx

rockstorm101, 2017-12-13:
Originally detected by @kliment: the spool manager makes pronsole depend on wx. I'm opening this issue to make everyone else aware of this.

Max-Headroom, 2017-12-24:
Traceback (most recent call last):
  File "pronterface.py", line 66, in <module>
  File "printrun\pronterface.pyc", line 2296, in __init__
  File "printrun\pronterface.pyc", line 220, in __init__
  File "printrun\pronterface.pyc", line 279, in reload_ui
  File "printrun\gui\__init__.pyc", line 155, in createTabbedGui
  File "printrun\gui\controls.pyc", line 299, in add_extra_controls
  File "printrun\gui\controls.pyc", line 126, in add
  File "wx\_core.pyc", line 14086, in Add
wx._core.PyAssertionError: C++ assertion "!CheckForIntersection(item)" failed at ..\..\src\common\gbsizer.cpp(225) in wxGridBagSizer::Add(): An item is already at that position

rockstorm101, 2017-12-27:
Hi, @Max-Headroom, how is this related to the spool manager?

commit d4944b48 (2017-12-30): Split Spool Manager's code into GUI and non-GUI

Separate the code from the Spool Manager that depends on wx, from the
one it does not, so pronsole does not depend on the wx module.

(Fixes #867)
closed 2017-12-30 by rockstorm101