kliment/Printrun #869: Monitoring 3-d printer and print jobs

mckatzenelson, 2017-12-29:
It would be helpful to have a capability to monitor, or view status, from another computer.

rockstorm101, 2018-01-07:
Hi, I haven't tried it but [according to the readme](https://github.com/kliment/Printrun#rpc-server) you might be able to ssh into the machine running pronterface/pronsole and query the status of the print job.

mckatzenelson, 2018-01-07:
Looks like the right idea, and perhaps it can be queried remotely. Will 
try it shortly.  Thank you

rockstorm101, 2018-01-07:
Please let us know your findings, this is an interesting feature.

mckatzenelson, 2018-01-09:
The code works on the same machine.  It did not work for another 
machine, "connection refused".

mckatzenelson, 2018-01-09:
Here is the string returned by the rpc server:

{'temps': {'B': ['80.3', '80.0'], 'T': ['185.0', '185.0']}, 'progress': 
1.9361282556918584, 'eta': [6093.301772092964, 7012.301772092964, 756], 
'z': 0.5, 'filename': 'some-file.gcode'}

Progress appears to be in percent.  I am not sure how to read 'eta', 
perhaps its the last coordinate?

mckatzenelson, 2018-01-09:
Found it. The 'eta' data seems to be seconds total, remaining, and elapsed.

rockstorm101, 2018-01-30:
Thanks a lot for sharing your findings.

mckatzenelson, 2018-02-21:
Attached is a python program to display the contents of the rpc status 
in readable form.  In case the attachment does not come through, the 
code is also listed inline.

Regarding the previous emails on this subject, the eta item turns out to 
be time remaining, total estimated time, both in seconds, and lines written

The procedure for using this is to copy the file to the machine where 
Pronterface will be running, set the permissions to allow execute, and 
then from another machine,

    $ ssh  ip-address  dirpath/pronterfacestatus.py

#!/usr/bin/python
'''
   pronterfacestatus.py

   Mitchell C. Nelson, Ph.D.
   Copyright 2017

   Version 0.0 alpha

'''

import xmlrpclib

def timestr( secs ):
     m, s = divmod(secs, 60)
     h, m = divmod(m, 60)
     return "%d:%02d:%02d" % (h, m, s)

rpc = xmlrpclib.ServerProxy('http://localhost:7978')
#print rpc.status()

filename = rpc.status()['filename']
temps = rpc.status()['temps']
bed=temps['B']
tip=temps['T']
progress = rpc.status()['progress']
eta = rpc.status()['eta']
z = rpc.status()['z']

print( '  %s'%filename )
print( '  bed: %s/%s extruder: %s/%s'%( bed[0],bed[1], tip[0],tip[1]) )
print( '  est: %s of %s remaining'%(timestr(eta[0]),timestr(eta[1])) )
print( '  printed %d lines, %.1f%% complete'%(eta[2],progress) )

kliment, 2018-03-24:
Would you be willing to contribute this code to Printrun as a pull request?