kliment/Printrun #870: pronterface not connecting

cdhmaui, 2017-12-31:
windows 7 enterprise
tried  drivers 1.0 and 1.1
smoothie shows in device manager with error
[image]
[image]

rockstorm101, 2017-12-31:
Hi @cdhmaui, could you share a screenshot of pronterface and the error you get when unable to connect?

cdhmaui, 2018-01-02:
[image]
 here is pronterface

rockstorm101, 2018-01-07:
Hi @cdhmaui, it seems to me like a Windows thing. Digging in google I've found:
 * A [Microsoft's answer](https://answers.microsoft.com/en-us/windows/forum/all/error-the-semaphore-timeout-period-has-expired/c139fd16-b0b5-4b1b-b5f5-161b6c801455) to a guy with the same error on a similar use case.
 * Some people suggest your port "COM7" might be already in use by another application which prevents pronterface from accessing it. A workaround [here](https://stackoverflow.com/a/6105595).

Please let us know if you solve this. Is any other application, such as the Arduino IDE, able to connect to your Smoothieboard through COM7?

rockstorm101, 2018-01-30:
Hi @cdhmaui, did any of the solutions suggested solve this issue?

cdhmaui, 2018-02-10:
I wasn’t able to resolve the issue I ended it by installing windows 10 pro while I like XP MS is ending its support they helped me with this issue but it remained unsolved time to move on the software evolution continues as always.
Iwant to thank you for your help and support in trying to resolve this issue in retrospect I thing that it might have been related to the fact that my version of XP was ENTERPRISE and I had some other issues after I retired and wasn’t linked to the company servers anymore.

rockstorm101, 2018-02-11:
Well, not the good news I was expecting :P. I'll close the issue since it does not seem like a bug in Printrun. Thanks for reaching back.

If anyone else encounters this problem again, please feel free to reopen.

closed 2018-02-11 by rockstorm101

genuisboy83, 2019-08-12:
It wont connect