kliment/Printrun #873: Not centering part on bed properly

mdroberts1243, 2018-01-09:
I have Printrun 1.6.0, Windows 10, Mendel printer with X:182, Y:200, Z:52 set in Options.
When loading a file it places the part off center, as if it thinks the X bed size is 200.
Using the Plater or Slic3r directly I see the part placed correctly... once Done in Plater the part is put offset on the print bed viewer in Printrun.  If I export gcode from Slic3r it is centered properly when loaded into Printrun.

rockstorm101, 2018-01-11:
HI @mdroberts1243, I need some help trying to reproduce this. Could you please state the steps to reproduce it one by one?

> When loading a file it places the part off center, as if it thinks the X bed size is 200.
> Using the Plater or Slic3r directly I see the part placed correctly... once Done in Plater the part is put offset on the print bed viewer in Printrun.

I guess in this cases you load an STL file?

> If I export gcode from Slic3r it is centered properly when loaded into Printrun.

and in this one you are loading the gcode file produced by Slic3r?

Thanks

rockstorm101, 2018-01-30:
Hi @mdroberts1243, were you able to solve this issue?

closed 2018-02-24 by rockstorm101