kliment/Printrun #874: failed to set power settings

DragonDon, 2018-01-14:
MacBook Air, 10.13.2

The print works but seeing the following error message through Pronterface:

[ERROR] Failed to set power settings:
Traceback (most recent call last):
  File "/Users/craigvanvliet/Desktop/PyInstaller-2.1/pronterface/build/pronterface/out00-PYZ.pyz/printrun.pronsole", line 1186, in startcb
  File "/Users/craigvanvliet/Desktop/PyInstaller-2.1/pronterface/build/pronterface/out00-PYZ.pyz/printrun.power", line 130, in powerset_print_start
  File "/Users/craigvanvliet/Desktop/PyInstaller-2.1/pronterface/build/pronterface/out00-PYZ.pyz/printrun.power.osx", line 75, in inhibit_sleep_osx
  File "/Users/craigvanvliet/Desktop/PyInstaller-2.1/pronterface/build/pronterface/out00-PYZ.pyz/printrun.power.osx", line 52, in AssertionCreateWithName
  File "/Users/craigvanvliet/Desktop/PyInstaller-2.1/pronterface/build/pronterface/out00-PYZ.pyz/printrun.power.osx", line 44, in StringToCFString
AttributeError: 'module' object has no attribute 'CFStringCreateWithCString'

chingling123, 2018-01-20:
Same problem to me...

Nocturtle, 2018-02-13:
This problem has also been reported in #749 & #619 .

kliment, 2018-03-23:
Closing as duplicate

closed 2018-03-23 by kliment

martinich, 2018-11-20:
This issue must be new as #619 was closed in 2015 and this is a binary from Nov 2017.

raould, 2019-09-02:
Printrun 1.6.0
I just downloaded this app today to my mac to run a new printer here in 2019, so this isn't some super old code on some super old laptop :-)

```
[ERROR] Failed to set power settings:
Traceback (most recent call last):
  File "/Users/craigvanvliet/Desktop/PyInstaller-2.1/pronterface/build/pronterface/out00-PYZ.pyz/printrun.pronsole", line 1193, in endcb
  File "/Users/craigvanvliet/Desktop/PyInstaller-2.1/pronterface/build/pronterface/out00-PYZ.pyz/printrun.power", line 134, in powerset_print_stop
  File "/Users/craigvanvliet/Desktop/PyInstaller-2.1/pronterface/build/pronterface/out00-PYZ.pyz/printrun.power.osx", line 82, in deinhibit_sleep_osx
AttributeError: 'function' object has no attribute 'assertion_id'
```