kliment/Printrun #875: gcode is not loading on pronterface

drewco83, 2018-01-14:
Hello I am running printrun febuary 2015 version. I am creating my models in fusion 360 CAD. My issue is when I load them into pronterface after slicing them just the initial outlines show up. Any thoughts
[image]
[image]

rockstorm101, 2018-01-15:
Hi, I assume you are using Windows? It seems that your gcode file is way too short.

 1. Could you please confirm whether this happens with the [latest release](https://github.com/kliment/Printrun/releases/latest)?

2. Could you please share both the STL you want to print and the gcode file you obtain from whatever slicing software you are using?

Thanks

drewco83, 2018-01-16:
Yes I am using windows 10 and pronterface version march 2012 right now. I downloaded the newest version yesterday but same issue. I attached requested files to the email. Also when I very first downloaded the program it let me covert another stl.file to gcode and worked just fine in pronterface but next day this is what happened.
thanks Drew

rockstorm101, 2018-01-16:
Hi,

 * I cannot find the files, please attach them in GitHub. (You might have to compress them into a .zip)

 * What slicing software are you using?

drewco83, 2018-01-16:
the version of slic3r that comes with the newest version of printrun

rockstorm101, 2018-01-30:
Hi @drewco83, were you able to solve this issue? If you won't share the files, finding a solution will be highly unlikely.

closed 2018-03-06 by rockstorm101