kliment/Printrun #876: how we can take bed and extruder temperature on printsole?

cagsurgunu, 2018-01-15:
I want to get temperature of bed and extruder from printsole? What are their codes.

Thank for help.

rockstorm101, 2018-01-15:
Hi, as is explained in the readme [[1]]:

> Type help to view the available commands. All commands have internal help, which you can access by typing "help commandname"

The command `gettemp` should give you both temperatures:
```
> help gettemp
Read the extruder and bed temperature.
```
If you wish to know the gcode command, it would me `M105` [[2]].

[1]: https://github.com/kliment/Printrun#using-pronsole
[2]: http://reprap.org/wiki/G-code#M105:_Get_Extruder_Temperature

cagsurgunu, 2018-01-15:
rosckstorm hi, i want to write a script on python idle. what is python codes for temp status. how can i read after p.sendnow(m105).