kliment/Printrun #877: Pronterface on Raspberry Pi 3

Fallen804, 2018-01-15:
Hello,
I just dont know how to install and run the pronterface on Rp the only thing i can do is to open the pronterface.py document but theres only code in it. My problem is i dont know which other programms i need to run Pronterface. Do i need Python ?
Please only respond in step by step im new on Raspberry pi :)
thx for the help!

rockstorm101, 2018-01-15:
Hi, what is the output of the command `lsb_release -a`?

rockstorm101, 2018-01-15:
Don't you get any more output after that? For instance, I get:
```
$ lsb_release -a
No LSB modules are available.
Distributor ID:	Raspbian
Description:	Raspbian GNU/Linux 8.0 (jessie)
Release:	8.0
Codename:	jessie
```
Anyway I just need to know whether you have Raspbian version 8 (_jessie_) or 9 (_strech_).

Fallen804, 2018-01-15:
I have 9 (stretch)
Thx for help :)

rockstorm101, 2018-01-15:
Good!, then installing Printrun should be as easy as:
```
sudo apt install printrun
```
After that, you should see pronterface and pronsole icons in the applications menu. If running non-GUI, simply type `pronsole` on the terminal to start pronsole.

Fallen804, 2018-01-16:
thanks the pronsole is working but how can i get this layout like on my PC? like on this picture:
[https://www.google.at/search?client=firefox-b&dcr=0&biw=1150&bih=656&tbm=isch&sa=1&ei=3iheWu_YAtCdsAfUz6KIDA&q=pronterface+raspberry&oq=pronterface+raspberry&gs_l=psy-ab.3..0i24k1.3269.5405.0.5727.10.2.0.8.8.0.185.306.0j2.2.0....0...1c.1.64.psy-ab..0.10.398...0.0.lEGFTQ4YWrY#imgrc=CTXTP_FcKzgBbM:](url)

rockstorm101, 2018-01-17:
What is seen in the picture is Pronterface not Pronsole.

rockstorm101, 2018-01-30:
The question seems solved so I'm closing this issue.

closed 2018-01-30 by rockstorm101

I-Dont-Remember, 2018-06-20:
Is the Raspbian package installation documented anywhere?  I didn't see it anywhere on the website or in the repo README, which led a coworker to spend a while trying to install from source before I found this issue.

rockstorm101, 2018-07-01:
Hi @I-Dont-Remember, Raspian derives from Debian, so the installation process should be ,if not identical, at least very similar to the one described for Debian.

belongumdave, 2020-04-06:
I have been following along, and now I have my pronterface running on the RPi.
However, I want to use the 7" screen, so it does not clutter up my 3D printer space. In so, how do I resize the 7" screen (from element14) so that I can access to the whole page. IE: to have some navigation bars to move around the page, anyone can help?