kliment/Printrun #879: pronterface not starting

vsolenoid, 2018-01-21:
Hi Kliment,

I managed to start Pronterface just once on my Win10 PC and it's not loading/starting after that.
I deleted the whole folder, downloaded and extracted Printrun-win-18Nov2017.zip again, but I still can't start the Pronterface. Any suggestions?

Regards,
Viktor Petkov

vsolenoid, 2018-01-21:
That's the log, if it helps:

Traceback (most recent call last):
  File "pronterface.py", line 66, in <module>
  File "printrun\pronterface.pyc", line 2296, in __init__
  File "printrun\pronterface.pyc", line 220, in __init__
  File "printrun\pronterface.pyc", line 279, in reload_ui
  File "printrun\gui\__init__.pyc", line 155, in createTabbedGui
  File "printrun\gui\controls.pyc", line 299, in add_extra_controls
  File "printrun\gui\controls.pyc", line 126, in add
  File "wx\_core.pyc", line 14086, in Add
wx._core.PyAssertionError: C++ assertion "!CheckForIntersection(item)" failed at ..\..\src\common\gbsizer.cpp(225) in wxGridBagSizer::Add(): An item is already at that position
Traceback (most recent call last):
  File "pronterface.py", line 66, in <module>
  File "printrun\pronterface.pyc", line 2296, in __init__
  File "printrun\pronterface.pyc", line 220, in __init__
  File "printrun\pronterface.pyc", line 279, in reload_ui
  File "printrun\gui\__init__.pyc", line 155, in createTabbedGui
  File "printrun\gui\controls.pyc", line 299, in add_extra_controls
  File "printrun\gui\controls.pyc", line 126, in add
  File "wx\_core.pyc", line 14086, in Add
wx._core.PyAssertionError: C++ assertion "!CheckForIntersection(item)" failed at ..\..\src\common\gbsizer.cpp(225) in wxGridBagSizer::Add(): An item is already at that position
Traceback (most recent call last):
  File "pronterface.py", line 66, in <module>
  File "printrun\pronterface.pyc", line 2296, in __init__
  File "printrun\pronterface.pyc", line 220, in __init__
  File "printrun\pronterface.pyc", line 279, in reload_ui
  File "printrun\gui\__init__.pyc", line 155, in createTabbedGui
  File "printrun\gui\controls.pyc", line 299, in add_extra_controls
  File "printrun\gui\controls.pyc", line 126, in add
  File "wx\_core.pyc", line 14086, in Add
wx._core.PyAssertionError: C++ assertion "!CheckForIntersection(item)" failed at ..\..\src\common\gbsizer.cpp(225) in wxGridBagSizer::Add(): An item is already at that position

rockstorm101, 2018-01-22:
Hi, could it be that you've changed from standard to tabbed UI mode?

Please check your configuration file. If there is a line saying "set uimode Tabbed" please change it back to "set uimode Standard". Tabbed mode does not currently work.

(The configuration file on Windows is typically found in " C:\Users\<yourusername>\printrunconf.ini")

vsolenoid, 2018-01-29:
Hey Rock,

That's exactly what happened! 
Thank you for the well described solution!

Cheers,
Viktor

 >-------- Оригинално писмо --------

 >От: Rock Storm notifications@github.com

 >Относно: Re: [kliment/Printrun] pronterface not starting (#879)

 >До: kliment/Printrun  

 >Изпратено на: 22.01.2018 09:43

 
 
  Hi, could it be that you've changed from standard to tabbed UI mode?
  
 
  
 Please check your configuration file. If there is a line saying "set uimode Tabbed" please change it back to "set uimode Standard". Tabbed mode does not currently work.
  
 
  
 (The configuration file on Windows is typically found in " C:\Users\ \printrunconf.ini")

rockstorm101, 2018-01-29:
Good to hear. Thanks for reporting back. Glad I could help. I'm closing the issue then.

closed 2018-01-29 by rockstorm101

mo106, 2018-03-21:
I had to reinstall windows to get the prontface running . It ran very well Till I whipped the winning horse by making some changes exactly as mentioned above.
But now I can not undo that as the prontface wont open at all , not even with the trouble shooter.
As i wanted to delete it wont as the process was open some where else aswell SO I installed the Process Explorere and there were almost 50 Prontface processes running , having killed all the only achievement was that The deletion became possibe.
Re install .but not loading // but the prontface exe text error begins .
Please advise how to , as installing windows again will increase a lot of work

kliment, 2018-03-21:
Did you delete your configuration?

mo106, 2018-03-21:
here are the pics

[image]

Traceback (most recent call last):
  File "pronterface.py", line 66, in <module>
  File "printrun\pronterface.pyc", line 2296, in __init__
  File "printrun\pronterface.pyc", line 220, in __init__
  File "printrun\pronterface.pyc", line 279, in reload_ui
  File "printrun\gui\__init__.pyc", line 155, in createTabbedGui
  File "printrun\gui\controls.pyc", line 299, in add_extra_controls
  File "printrun\gui\controls.pyc", line 126, in add
  File "wx\_core.pyc", line 14086, in Add
wx._core.PyAssertionError: C++ assertion "!CheckForIntersection(item)" failed at ..\..\src\common\gbsizer.cpp(225) in wxGridBagSizer::Add(): An item is already at that position
Traceback (most recent call last):
  File "pronterface.py", line 66, in <module>
  File "printrun\pronterface.pyc", line 2296, in __init__
  File "printrun\pronterface.pyc", line 220, in __init__
  File "printrun\pronterface.pyc", line 279, in reload_ui
  File "printrun\gui\__init__.pyc", line 155, in createTabbedGui
  File "printrun\gui\controls.pyc", line 299, in add_extra_controls
  File "printrun\gui\controls.pyc", line 126, in add
  File "wx\_core.pyc", line 14086, in Add
wx._core.PyAssertionError: C++ assertion "!CheckForIntersection(item)" failed at ..\..\src\common\gbsizer.cpp(225) in wxGridBagSizer::Add(): An item is already at that position
Traceback (most recent call last):
  File "pronterface.py", line 66, in <module>
  File "printrun\pronterface.pyc", line 2296, in __init__
  File "printrun\pronterface.pyc", line 220, in __init__
  File "printrun\pronterface.pyc", line 279, in reload_ui
  File "printrun\gui\__init__.pyc", line 155, in createTabbedGui
  File "printrun\gui\controls.pyc", line 299, in add_extra_controls
  File "printrun\gui\controls.pyc", line 126, in add
  File "wx\_core.pyc", line 14086, in Add
wx._core.PyAssertionError: C++ assertion "!CheckForIntersection(item)" failed at ..\..\src\common\gbsizer.cpp(225) in wxGridBagSizer::Add(): An item is already at that position

mo106, 2018-03-21:
Sir 
where and how do I delete the config

kliment, 2018-03-21:
Go to c:\users\(yourusername)\AppData and check the subfolders for files named pronsolerc or printrunconf.ini and delete those. If you can't find them search your users folder for those names.

mo106, 2018-03-21:
thanks I will do that and report back

mo106, 2018-03-21:
Yes I am IN. thank It has resolved a matter on permanent basis and the program has recovered 
Now my second issue.
1.Is it possible to have the machine coordinates display while the machine Runs , 
2.How to set the limits >> where exactly should be the machine on HOME and how to do that
Please advise

kliment, 2018-03-21:
M114 command will show you the machine coordinates
Limits are set in two places - in your firmware, and in the options (printer settings). Clamp manual moves option will enforce these limits. When homed your machine should be at whatever you set to home position in the options. Unfortunately I will not be able to help you with configuring your firmware and printer mechanics. Also, please do not add new questions to existing issues. If you have a new issue, open a new issue.

mo106, 2018-03-21:
[image]

mo106, 2018-03-21:
Thank you My issues are almost solved
Great work The software is just awesome
Do you have a tutorial On the internet Please let me know 
Thanks

Praktiker, 2018-04-02:
Hi I have the same problem on the Mac. It works from one profile but not from my standard profile anymore. It used to work, but I had to kill the task since printing went haywire. After that printrun (Version Nov 2018) does not start anymore. Where can I find the config on a mac?

kliment, 2018-04-02:
Hey @Praktiker , the easiest way to resolve this is to upgrade to the latest release candidate at https://github.com/kliment/Printrun/releases

Praktiker, 2018-04-02:
Thanks I will try that. Still, is there a way to change configuration without the gui? (config-file)

kliment, 2018-04-02:
@Praktiker on osx the configuration file will be either in your home directory (cd ~ in terminal to find it) or in  ~/Library/Application Support/printrun

kliment, 2018-04-02:
@Praktiker you can also search for .pronsolerc files using spotlight

Praktiker, 2018-04-02:
Though, I couldn't find the folders or files on my mac... (I was looking for them before anyway). No problem it works, so I am fine. Have a nice day.

Spideychicken, 2019-11-10:
I believe I'm having the same problems as stated above, but my config file or any other files related to pronterface aren't anywhere to be found outside of the folder I unzipped it to.
And just to be clear I have already checked in C\Users\"name"\
I have checked my entire disk and it is nowhere to be found.

Heres my log
Traceback (most recent call last):
  File "pronterface.py", line 66, in <module>
  File "printrun\pronterface.pyc", line 2296, in __init__
  File "printrun\pronterface.pyc", line 220, in __init__
  File "printrun\pronterface.pyc", line 279, in reload_ui
  File "printrun\gui\__init__.pyc", line 155, in createTabbedGui
  File "printrun\gui\controls.pyc", line 299, in add_extra_controls
  File "printrun\gui\controls.pyc", line 126, in add
  File "wx\_core.pyc", line 14086, in Add
wx._core.PyAssertionError: C++ assertion "!CheckForIntersection(item)" failed at ..\..\src\common\gbsizer.cpp(225) in wxGridBagSizer::Add(): An item is already at that position