kliment/Printrun #880: A question, not an issue.

RichardNg, 2018-02-04:
Hello, I am not really reporting an issue but rather a question I don't know where to ask.
I switched to a Raspberry Pi for my pronterface. This pi has a 3.5" LCD. I hack the source code to make the ui look nicely on the 3.5" LCD. My last problem is that the Extrude. Reverse and length elements, they are way below on the bottom left.
[image] And the extrude length box is obscured unless I move the window up. unless I move the window up, even then they are not semmetrical on the entire UI.  Ideally I would like it looks like this: 
[image]. I dug into these two files try to modify the controls.py and pronterface.py file, it seems like the original use some sort of array or something, I can't figure out how the panel, gridboxsier are layout,( I am pretty much a newbie to python, I figure this is a good learning exercise. ) :-) I wonder if someone has more python experience can give me some pointers ?Thanks.

kliment, 2018-03-24:
Yeah, it's a gridbag sizer, so you have a grid and you place objects on specific coordinates and each object has a size in number of cells. The objects you are trying to move are edist_label, edist_val and edist_unit for the extrude length, and extrude and reverse for the buttons. Edit their position and span mapping values to move them around. Make sure they don't step on top of other objects (it's easiest if you draw the grid on paper and mark out the coordinates taken up by each object)

kliment, 2018-03-25:
Also, you might want to switch to the mini controls (settings->options->user interface->control mode)

kliment, 2018-03-25:
I'm going to close this issue as your question has been answered. Feel free to comment again if you need further help.

closed 2018-03-25 by kliment

RichardNg, 2018-03-25:
Thank you very for the reply. I have now completed the proterface UI. :)