kliment/Printrun #881: Cannot tell if print is active with printcore

slawlor6, 2018-02-07:
I am working on a sorter that uses x-y stepper motors to scan an object in the input bin and place it in a destination bin. I am able to get my setup working really well for objects one at a time, but if I automate the loading I can't figure out when the x-y steppers have come back from the destination to the input. I thought if I used the example, p.startprint(gcode), I would know if a print was active or not based on the comment from printcore.py below:

 def startprint(self, gcode, startindex = 0):
        """Start a print, gcode is an array of gcode commands.
        returns True on success, False if already printing.
        The print queue will be replaced with the contents of the data array,
        the next line will be set to 0 and the firmware notified. Printing
        will then start in a parallel thread.
        """

If I startprint() while the other print is running, the gcode is just appended to the previous run without hesitation. I need to do some sensing in between runs. How can I tell if the print has completed by using printcore? I don't want to use pronsole or pronterface.

slawlor6, 2018-02-16:
I am looking into the endcb() function in the event handler tree, but have not been successful. Do you believe this will work in a python implementation so that I can identify when a batch of gcode sent to the printer has been executed?

kliment, 2018-03-24:
You could also check (assuming your printcore object is named p) the following condition:

`while p.printing: #don't call startprint yet`

closed 2018-03-24 by kliment