kliment/Printrun #888: Please support the T? G-code

bubnikv, 2018-02-13:
Hello.

Our Prusa i3 MK2 Multi-Material firmware based on the Marlin firmware understands a new G-code: T? instructs the firmware to ask user to select the spool to print from. We have an error report, that pronterface does not understand the T? G-code and it refuses to print.

https://github.com/prusa3d/Slic3r/issues/680

Would you please add a support for the new T? G-code? 
Thank you.

rockstorm101, 2018-02-14:
Hi, could you please provide a simple example gcode file using command "T?" so we can play around with it?

Thanks

bubnikv, 2018-02-14:
Sure. Here is an example, generated with the Slic3r Prusa Edition for the "Original Prusa i3 MK2 MM Single Mode" Printer profile.

[20mmbox.zip](https://github.com/kliment/Printrun/files/1725636/20mmbox.zip)

Thanks for your attention.

kliment, 2018-03-19:
Hey, this was a two line change, I implemented it in both master and the 2.x (py3/wxp4) branch. @bubnikv, can you verify this works for you?

FidelCapo, 2018-03-20:
@kliment I tested it on our multi-printer printer and everything went well. Thanks for your implementation.

kliment, 2018-03-20:
Awesome. If you're in a position to do so, testing of the 2.x branch would be greatly appreciated (not just for this issue, but also in general). I'm closing this issue as resolved.

closed 2018-03-20 by kliment

bubnikv, 2018-03-20:
@kliment Thanks, we will do a through testing before putting the current
version into our drivers package.

FidelCapo, 2018-03-20:
@klimentI Would it be possible to provide a compiled last version of pronterface for win? Thanks

kliment, 2018-03-20:
Only after the 2.0 release is done. I need to redo the windows build/packaging process.

FidelCapo, 2018-03-20:
OK, Thank you

kliment, 2018-03-23:
I made the slic3r integration in printrun detect and handle slic3rPE config files (for whatever reason they're in a separate location from normal slic3r configs, and I was testing how slic3r integration deals with missing config files, and noticed this). I have a binary build setup now for Windows and OSX, so I'll probably have something for you to test within the next couple weeks.

FidelCapo, 2018-03-28:
So do you have any build for windows and OSX that you could send me?

rockstorm101, 2018-03-28:
@FidelCapo please see https://github.com/kliment/Printrun/releases/tag/printrun-2.0.0rc5

FidelCapo, 2018-03-28:
Oh, sorry, I did not find it. Thank you

FidelCapo, 2018-03-28:
I tested prints on win Pronterface build on MK2, MK3, MK2MM and MK2MM Single mode and all are ok. Error with command T in MM Single mode gcode did not reappear.

FidelCapo, 2018-03-28:
Do you have any estimate of when you release final version? If you want, I can help with testing because I'm going to test Pronterface for us anyway.

hroncok, 2018-03-28:
I suppose when no new bug will be opened for a ~week. Also #913 is open.

bubnikv, 2018-03-28:
@hroncok 
> I suppose when no new bug will be opened for a ~week.
Don't be so harsh to yourself, look at the list of open issues at both slic3rs. With such a policy we would not make a release ever :-)

hroncok, 2018-03-28:
Well here the traffic isn't that high. I suppose when no new regressions are reported was what I intended to say.