kliment/Printrun #889: object not centered on plater

cdhmaui, 2018-02-22:
[image]

after loading file object shows up off the plater ?

kliment, 2018-02-22:
That's exciting, can you post the gcode file and the dimensions your bed is set to?

cdhmaui, 2018-02-22:
the STL files look good in Slic 3 then export G code file then when loaded into pronterface is when the shift happens every file ive printed 20+ does the same thing i have a 300mm bed and set it at 250 as that is the heat pad size it round as in pic

rockstorm101, 2018-02-22:
I'm not sure why did this issue got closed, but it seems OP won't share any files so I guess is fine.
If anyone else encounters the same problem, please make sure not only build dimensions are correct but also the home positions.

cdhmaui, 2018-02-23:
[image]
[image]
here are the screens showing plater image offset

rockstorm101, 2018-02-23:
Hi, could you please share a screenshot of your build dimensions configuration? (A screenshot of the window that opens from _Settings_>_Options_, the "Printer settings" tab)

cdhmaui, 2018-02-23:
[image]

rockstorm101, 2018-02-24:
Hi, try setting:
 * _X offset_ = -100.00
 * _Y offset_ = -100.00

This should center your parts in the 3D view. Please let us know if it works.

codern00b, 2018-02-28:
The part not being on the plate has been my experience too. I am also on 1.6.0 (windows 7).
When I moved the part to the middle of the delta bed, then the effector crashed into the side of the frame, as it tried to print outside the frame. I now just ignore the display issue, as it prints just fine, but since I saw this thread, I thought I will add my experience so that it can be seen it is not an isolated incident.

rockstorm101, 2018-02-28:
Hi @codern00b, have you tried tweaking the offsets as I suggested? Could that work?

cdhmaui, 2018-03-01:
I tried the offsets it moved the object to the center but it moved my tower calibration points totally off the bed so I put it back as before I won’t be able to post on the site until I return from NASCAR race in Las Vegas

kliment, 2018-03-22:
@rockstorm101 could you reproduce this on your setup with the offset setting as you suggested?

rockstorm101, 2018-03-22:
Setting the offsets as I suggested does place the parts on the center of a circular build plate. However my printer has a squared build plate and I do calibration directly from firmware. Truth be told, I have no real way to test the reported problem. But my guess is setting the offsets like that does offset more than just the parts displayed on the 3D view.

The offset thing is just a workaround. Which is necessary only because currently Printrun sets the origin for both rectangular and circular on the left bottom corner. For circular beds the origin should be the center of the build plate.