kliment/Printrun #895: Segmentation fault when changing interface mode

rockstorm101, 2018-03-02:
* OS: Debian 10 Buster
 * Version: 2.0.0rc2

I get a "Segmentation fault" when changing the interface mode. Switching from 'Standard' to 'Compact' interface modes produces:
```
Failed to run callback after setting "uimode":
Traceback (most recent call last):
  File "/home/rock/Downloads/Printrun/printrun/settings.py", line 352, in _set
    if cb is not None: cb(key, value)
  File "/home/rock/Downloads/Printrun/printrun/pronterface.py", line 282, in reload_ui
    self.splitterwindow.SetSashPosition(self.settings.last_sash_position)
RuntimeError: wrapped C/C++ object of type SplitterWindow has been deleted

Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/wx/core.py", line 3190, in <lambda>
    lambda event: event.callable(*event.args, **event.kw) )
RuntimeError: wrapped C/C++ object of type GcodeViewPanel has been deleted
```

hroncok, 2018-03-03:
Reproduced.

hroncok, 2018-03-05:
When unchecking *display temperature graph* and clicking OK, I also get Segmentation fault.

hroncok, 2018-03-05:
Reproduced on macOS.

kliment, 2018-03-19:
Okay, disabling monitoring seems to make all segfaults go away. So clearly the monitoring timer is sending events to nonexistent objects for some reason

kliment, 2018-03-19:
I think i got this with  61581e1ec2dbb32bd3e237cbbbdbe3d1e73aab8d - can anyone test?

hroncok, 2018-03-19:
On it.

hroncok, 2018-03-19:
Confirmed fixed on Linux.

hroncok, 2018-03-19:
No segfault on macOS either.

closed 2018-03-19 by hroncok