kliment/Printrun #897: Some numeric options are compacted down to basically 0 length, some sliders are too short

VanessaE, 2018-03-03:
Using 2.0, commit 116fddaebb48adccb91ee23b34c26bb09ba27797, under Debian 9.3 stable

[image]
(The build dimensions, in particular)
[image]

Here's what it looked like before (minus one control that didn't exist when the screenshot was taken, please ignore the circled item):

[image]

<details>
    <summary>All the gtk-related spew from the terminal</summary>

```
(pronterface.py:11045): Gtk-CRITICAL **: gtk_box_gadget_distribute: assertion 'size >= 0' failed in GtkSpinButton

(pronterface.py:11045): Gtk-CRITICAL **: gtk_box_gadget_distribute: assertion 'size >= 0' failed in GtkSpinButton

(pronterface.py:11045): Gtk-WARNING **: Negative content width -5 (allocation 13, extents 9x9) while allocating gadget (node entry, owner GtkEntry)

(pronterface.py:11045): Gtk-CRITICAL **: gtk_box_gadget_distribute: assertion 'size >= 0' failed in GtkSpinButton

(pronterface.py:11045): Gtk-WARNING **: Negative content width -17 (allocation 1, extents 9x9) while allocating gadget (node entry, owner GtkEntry)

(pronterface.py:11045): Gtk-WARNING **: Negative content width -12 (allocation 22, extents 17x17) while allocating gadget (node button, owner GtkButton)

(pronterface.py:11045): Gtk-WARNING **: Negative content width -15 (allocation 19, extents 17x17) while allocating gadget (node button, owner GtkButton)

(pronterface.py:11045): Gtk-CRITICAL **: gtk_box_gadget_distribute: assertion 'size >= 0' failed in GtkSpinButton

(pronterface.py:11045): Gtk-CRITICAL **: gtk_box_gadget_distribute: assertion 'size >= 0' failed in GtkSpinButton

(pronterface.py:11045): Gtk-WARNING **: Negative content width -5 (allocation 13, extents 9x9) while allocating gadget (node entry, owner GtkEntry)

(pronterface.py:11045): Gtk-CRITICAL **: gtk_box_gadget_distribute: assertion 'size >= 0' failed in GtkSpinButton

(pronterface.py:11045): Gtk-WARNING **: Negative content width -17 (allocation 1, extents 9x9) while allocating gadget (node entry, owner GtkEntry)

(pronterface.py:11045): Gtk-WARNING **: Negative content width -12 (allocation 22, extents 17x17) while allocating gadget (node button, owner GtkButton)

(pronterface.py:11045): Gtk-WARNING **: Negative content width -15 (allocation 19, extents 17x17) while allocating gadget (node button, owner GtkButton)

(pronterface.py:11045): Gtk-CRITICAL **: gtk_box_gadget_distribute: assertion 'size >= 0' failed in GtkSpinButton

(pronterface.py:11045): Gtk-CRITICAL **: gtk_box_gadget_distribute: assertion 'size >= 0' failed in GtkSpinButton

(pronterface.py:11045): Gtk-WARNING **: Negative content width -5 (allocation 13, extents 9x9) while allocating gadget (node entry, owner GtkEntry)

(pronterface.py:11045): Gtk-CRITICAL **: gtk_box_gadget_distribute: assertion 'size >= 0' failed in GtkSpinButton

(pronterface.py:11045): Gtk-WARNING **: Negative content width -17 (allocation 1, extents 9x9) while allocating gadget (node entry, owner GtkEntry)

(pronterface.py:11045): Gtk-WARNING **: Negative content width -12 (allocation 22, extents 17x17) while allocating gadget (node button, owner GtkButton)

(pronterface.py:11045): Gtk-WARNING **: Negative content width -15 (allocation 19, extents 17x17) while allocating gadget (node button, owner GtkButton)

(pronterface.py:11045): Gtk-CRITICAL **: gtk_box_gadget_distribute: assertion 'size >= 0' failed in GtkSpinButton

(pronterface.py:11045): Gtk-CRITICAL **: gtk_box_gadget_distribute: assertion 'size >= 0' failed in GtkSpinButton

(pronterface.py:11045): Gtk-WARNING **: Negative content width -5 (allocation 13, extents 9x9) while allocating gadget (node entry, owner GtkEntry)

(pronterface.py:11045): Gtk-CRITICAL **: gtk_box_gadget_distribute: assertion 'size >= 0' failed in GtkSpinButton

(pronterface.py:11045): Gtk-WARNING **: Negative content width -17 (allocation 1, extents 9x9) while allocating gadget (node entry, owner GtkEntry)

(pronterface.py:11045): Gtk-WARNING **: Negative content width -12 (allocation 22, extents 17x17) while allocating gadget (node button, owner GtkButton)

(pronterface.py:11045): Gtk-WARNING **: Negative content width -15 (allocation 19, extents 17x17) while allocating gadget (node button, owner GtkButton)

[... 255 lines cut ...]
python3 -m venv v3
. ./v3/bin/activate
pip3 install --upgrade pip
pip3 install -U -f https://extras.wxpython.org/wxPython4/extras/linux/gtk3/debian-9/wxPython-4.0.1-cp35-cp35m-linux_x86_64.whl wxPython
cp -R /usr/lib/python3/dist-packages/gi \
	/usr/lib/python3/dist-packages/cairo \
	/usr/lib/python3/dist-packages/cairocffi \
	/usr/lib/python3/dist-packages/cairosvg \
	    v3/lib/python3.5/site-packages/

cat requirements.txt \
	|grep -v pygobject \
	|grep -v cairosvg \
	     > requirements-modified.txt
pip install wheel pkgconfig Cython
pip install -r requirements-modified.txt
python3 setup.py build_ext --inplace

# Run Pronterface like so:
#
cd /home/vanessa/RepRap/Printrun/
. ./v3/bin/activate
python3 pronterface.py
```
</details>

kliment, 2018-03-03:
Spinbox issue reproduced on Ubuntu 16.04 LTS using same (virtualenv) install method

hroncok, 2018-03-05:
We have some gtk3 patch in Fedora, I'm preping a PR. It might help.

kliment, 2018-03-05:
It appears gtk3 changed the default gtkspinbutton style to make it inconsistent with other platforms, including gtk2, and to have the controls take up way more space. http://k35.kn3.net/taringa/4/5/9/0/2/8/1/marianxs/F8B.jpg I don't expect there's any way to fix that short of a change to gtk3 itself

hroncok, 2018-03-05:
Could you please check with https://github.com/kliment/Printrun/pull/899?

VanessaE, 2018-03-05:
In case #899 won't do, is it possible to use a simple numeric field, and separate, dedicated buttons next to it that can change it, without using an actual spinbutton?

kliment, 2018-03-05:
Not super happy with the result but at least it can be used. Ugh, gtk3 design decisions.

closed 2018-03-05 by kliment