kliment/Printrun #900: 2.x: macOS install issues

hroncok, 2018-03-05:
The instructions for macOS should cover this:

 * **dbus should be Linux only dependency:** there's probably no no dbus on macOS, so dbus-python doesn't install. I've tried to install dbus via brew, but ti didn't help much. also dbus is used fox linux-ish screensaver detection
 * **pygobject wanted gobject-introspection and libffi pkgconfig:** solved by `brew install gobject-introspection` (pulls in `libffi`) and installing dependencies with `PKG_CONFIG_PATH=/usr/local/opt/libffi/lib/pkgconfig`

kliment, 2018-03-05:
can you expand on the last point - what did you need to do exactly to make it work?

hroncok, 2018-03-05:
`PKG_CONFIG_PATH=/usr/local/opt/libffi/lib/pkgconfi pip install -r requirements.txt`

BTW (slowly adding more issue to the top)

hroncok, 2018-03-05:
I was also hit by http://trac.wxwidgets.org/ticket/16555 but that's probably just because I virtualize the macOS.

hroncok, 2018-03-05:
One way would be to have `requirements.linux.txt` with:

```
-r requirements.txt
dbus-python
```

And `requirements.macos.txt` with:

```
-r requirements.txt
pyobjc-core
pyobjc-framework-Cocoa
```

We could then also have `requirements.windows.txt` with just:

```
-r requirements.txt
```

The otehr way si to mention those dependencies in the instructions only.

kliment, 2018-03-05:
This is a great idea - testing on windows I noticed we need to make some changes to requirements.txt there too - at least removing dbus. I need to investigate what we're using pygobject for, I think that might be linux-specific too.

kliment, 2018-03-05:
Hmm, as far as I can tell we're not using pygobject for anything anymore - I can't find any gobject or gi imports

hroncok, 2018-03-05:
That should make it easier :)

rockstorm101, 2018-03-05:
I believe It is possible to have a single requirements file for all platforms. According to [pip's documentation](https://pip.pypa.io/en/stable/reference/pip_install/#requirement-specifiers), we could specify what platform a dependency is for. E.g.:
```
pyreadline ; sys_platform == 'win32'
dbus-python >= 1.2.0 ; sys_platform == 'linux'
```
I've always had the suspicion that some of those requirements were no longer really needed.

hroncok, 2018-03-05:
Oh, I thought this was only possible in setuptools install_requires.

hroncok, 2018-03-07:
Confirmed.

closed 2018-03-07 by hroncok