kliment/Printrun #902: How to use pronsole commands in a shell script??

sebascarra, 2018-03-07:
Hello, I need to run some pronsole commands from a bash script but I have found no way to use pronsole in a non-interactive manner. For example, I would like to do something like this:

pronsole < connect 192.168.3.222:23
pronsole < G28
pronsole < M105 #get temperature information on stdout

How can I do something like this?? Pronsole seems to be made for a human to use but there's no way to take advantage of it from another sheel as far as I can tell...

This problem applies to many other programs, such as wpa_cli (which fortunately has a non-interactive mode to allow shell scripting with it).

I need a workaround. Thanks for your help!!

slick8086, 2018-03-08:
I'm new to pronterface, but I looked at the source of pronsole.py and it has an -e --execute option.  That probably does what you want.

kliment, 2018-03-08:
> I'm new to pronterface, but I looked at the source of pronsole.py and it
> has an -e --execute option. That probably does what you want.
> 

Almost - the problem is that the connection state will not persist
between calls, so you can't call it with connect, then call it again
with a command - as soon as the first one terminates the connection will
be closed. So the example given above would not work. Other than that,
yes, that is the correct way to do it.

sebascarra, 2018-03-14:
I ended up exploring this: https://en.wikipedia.org/wiki/Expect
Using that will use pronsole's interactive mode as if it were a human injecting commands! No need to use the --execute option unless you only need to do simple stuff.

kliment, 2018-03-19:
That's a good solution if you need persistent state. Thanks for suggesting it. I consider this issue resolved now, but feel free to reopen if you feel otherwise.

closed 2018-03-19 by kliment

sebascarra, 2022-07-07:
> > I ended up exploring this: https://en.wikipedia.org/wiki/Expect Using that will use pronsole's interactive mode as if it were a human injecting commands! No need to use the --execute option unless you only need to do simple stuff.
> 
> @sebascarra did it work ? I tried using expect to do that but for some reason it never seemed to work

Hey!! It's been almost 4 years since this work. Back in the day I had very little Python experience, so I was afraid of tinkering with Pronsole's code. If you're a programmer, I would just rather implement this feature myself, writing proper Python code instead of bothering with Expect. I don't remember what I ended up doing with Expect but I do remember it didn't work very well, so I ended up scrapping it and shortly after I left the company where I was playing around with this, so I never ended up with a proper solution. Like I said, with the experience I have today I would judt implement this modifying Pronsole's source code. Good luck!!

sebascarra, 2022-07-07:
@meutzitzu CC just in case my previous comment didn't notify you.

volconst, 2022-07-07:
@meutzitzu What about multiple -e options like this https://github.com/kliment/Printrun/issues/1263#issuecomment-1133586644