kliment/Printrun #903: Menu icon incorrectly changing after failed .obj change

siorai, 2018-03-08:
Branch: 2.x 
Commit b33d030 
Environment: Win10, version 1803, OS Build 17112.1

[image]

Seems that when I select the Load icon, and attempt to load a .obj file, it fails ( since I don't have a slicer installed, that's to be expected ) however instead of continuing to display 'load' it switches to 'cancel' as if it's been loaded.  Selecting the 'cancel' button also seems to perform the same functionality as if it were still the load button. 

Steps to reproduce: 

install via pip3 install -r requirements.txt
python3 pronterface.py 
select load from main UI, load a *.obj file
observer change in load button change to 'cancel'

Can make available the 2 .obj files I've tried if needed.

rockstorm101, 2018-03-08:
I can confirm this issue in Debian. It is not a regression, this behaviour dates back to at least version 1.6.0.

kliment, 2018-03-22:
Should be fixed in 2.x now - can someone confirm?

hroncok, 2018-03-22:
Loading a bogus file won't change the button to Cancel now.

Loading a good file does not change the button to Cancel either.

When is the button **supposed** to say Cancel?

kliment, 2018-03-22:
While a file is being sliced and loaded, until the first layer has been processed, so if your slicer is slow to process a huge file you can stop it.

hroncok, 2018-03-22:
OK. That happens.

closed 2018-03-22 by hroncok