kliment/Printrun #905: Initial popup size for Tools -> Spool Manager could stand to be a little bigger

siorai, 2018-03-08:
Branch: 2.x
Commit b33d030
Environment: Win10, version 1803, OS Build 17112.1

Prior to loading a spool:
[image]

After loading a spool:
[image]

It's worth mentioning that after a spool is loaded, as the manager is closed and reopened, the window displayed is properly sized:

[image]

Steps to reproduce:

install via pip3 install -r requirements.txt
python3 pronterface.py
select Tools -> Spool Manager

siorai, 2018-03-08:
Side note: the icon for the Spool Manager could stand to be something other than the standard icon for win32 applications as well.

kliment, 2018-03-08:
Similar issue with the new macro dialog - buttons get cut off 
[image]

rockstorm101, 2018-03-09:
I'll look into this. If anyone wants to beat me to it, please go ahead.

kliment, 2018-03-19:
Spool manager now looks decent for me - @siorai, can you test on your setup?

hroncok, 2018-03-19:
That's my spoolmanager as I open it:

[image]

siorai, 2018-03-22:
Looks great!

closed 2018-03-22 by siorai