kliment/Printrun #907: Editing/re-defining a macro causes duplicates to appear in the menu

VanessaE, 2018-03-10:
At commit eb6bd431cf7c9057492e0c0f011590f8fd16a8cc 

Define two or more macros
Close Pronterface and restart
Edit one of the macros and save
Check "Settings -> Macros" and you'll find two copies of each defined macro (i.e. including ones you didn't re-edit).

hroncok, 2018-03-20:
Reproduced on 610f9ff. This is the log from terminal:

Created macro foo:

```
Macro 'foo' defined
Macro 'foo' defined
```

Created macro bar:

```
Macro 'bar' defined
Macro 'bar' defined
```

Restarted, edited macro foo:

```
Macro 'foo' defined
Macro 'foo' defined
```

kliment, 2018-03-22:
Fixed in latest commit. @VanessaE please confirm it works correctly now.

VanessaE, 2018-03-22:
Seems to be solved.

commit e30e0df9 (2018-03-22): Change DeleteItem to DestroyItem to remove entries from menu. Fixes #907
closed 2018-03-22 by VanessaE