kliment/Printrun #909: Opening, closing the large view of the temperature history graph causes segfault

VanessaE, 2018-03-10:
```bash
vanessa@rainbird:~$ cat Scripts/run-pronterface-2.x.sh 
#!/bin/bash

cd /home/vanessa/RepRap/Printrun-2.x/
. ./v3/bin/activate
python3 pronterface.py -c /home/vanessa/.pronsolerc-2.x
```

```
vanessa@rainbird:~$ Scripts/run-pronterface-2.x.sh 
Fontconfig warning: "/etc/fonts/conf.d/50-user.conf", line 14: reading configurations from ~/.fonts.conf is deprecated. please move it to /home/vanessa/.config/fontconfig/fonts.conf manually
Generating grammar tables from /usr/lib/python3.5/lib2to3/Grammar.txt
Generating grammar tables from /usr/lib/python3.5/lib2to3/PatternGrammar.txt
07:33:02: Debug: Unsupported OpenGL attribute 1631008256
07:33:02: Debug: Unsupported OpenGL attribute 21905
07:33:02: Debug: Unsupported OpenGL attribute 73

(pronterface.py:662): Gtk-CRITICAL **: gtk_box_gadget_distribute: assertion 'size >= 0' failed in GtkSpinButton

(pronterface.py:662): Gtk-CRITICAL **: gtk_box_gadget_distribute: assertion 'size >= 0' failed in GtkSpinButton

(pronterface.py:662): Gtk-CRITICAL **: gtk_box_gadget_distribute: assertion 'size >= 0' failed in GtkSpinButton

(pronterface.py:662): Gtk-CRITICAL **: gtk_box_gadget_distribute: assertion 'size >= 0' failed in GtkSpinButton

(pronterface.py:662): Gtk-CRITICAL **: gtk_box_gadget_distribute: assertion 'size >= 0' failed in GtkSpinButton

(pronterface.py:662): Gtk-CRITICAL **: gtk_box_gadget_distribute: assertion 'size >= 0' failed in GtkSpinButton

(pronterface.py:662): Gtk-WARNING **: Negative content width -12 (allocation 22, extents 17x17) while allocating gadget (node button, owner GtkButton)

(pronterface.py:662): Gtk-WARNING **: Negative content width -15 (allocation 19, extents 17x17) while allocating gadget (node button, owner GtkButton)

(pronterface.py:662): Gtk-CRITICAL **: gtk_box_gadget_distribute: assertion 'size >= 0' failed in GtkSpinButton

(pronterface.py:662): Gtk-CRITICAL **: gtk_box_gadget_distribute: assertion 'size >= 0' failed in GtkSpinButton

(pronterface.py:662): Gtk-CRITICAL **: gtk_box_gadget_distribute: assertion 'size >= 0' failed in GtkSpinButton

(pronterface.py:662): Gtk-CRITICAL **: gtk_box_gadget_distribute: assertion 'size >= 0' failed in GtkSpinButton

(pronterface.py:662): Gtk-CRITICAL **: gtk_box_gadget_distribute: assertion 'size >= 0' failed in GtkSpinButton

(pronterface.py:662): Gtk-CRITICAL **: gtk_box_gadget_distribute: assertion 'size >= 0' failed in GtkSpinButton

(pronterface.py:662): Gtk-WARNING **: Negative content width -12 (allocation 22, extents 17x17) while allocating gadget (node button, owner GtkButton)

(pronterface.py:662): Gtk-CRITICAL **: gtk_box_gadget_distribute: assertion 'size >= 0' failed in GtkSpinButton

(pronterface.py:662): Gtk-CRITICAL **: gtk_box_gadget_distribute: assertion 'size >= 0' failed in GtkSpinButton

(pronterface.py:662): Gtk-CRITICAL **: gtk_box_gadget_distribute: assertion 'size >= 0' failed in GtkSpinButton

(pronterface.py:662): Gtk-CRITICAL **: gtk_box_gadget_distribute: assertion 'size >= 0' failed in GtkSpinButton

(pronterface.py:662): Gtk-CRITICAL **: gtk_box_gadget_distribute: assertion 'size >= 0' failed in GtkSpinButton

(pronterface.py:662): Gtk-CRITICAL **: gtk_box_gadget_distribute: assertion 'size >= 0' failed in GtkSpinButton

(pronterface.py:662): Gtk-WARNING **: Negative content width -12 (allocation 22, extents 17x17) while allocating gadget (node button, owner GtkButton)

(pronterface.py:662): Gtk-WARNING **: Negative content width -15 (allocation 19, extents 17x17) while allocating gadget (node button, owner GtkButton)

(pronterface.py:662): Gtk-CRITICAL **: gtk_box_gadget_distribute: assertion 'size >= 0' failed in GtkSpinButton

(pronterface.py:662): Gtk-CRITICAL **: gtk_box_gadget_distribute: assertion 'size >= 0' failed in GtkSpinButton

(pronterface.py:662): Gtk-CRITICAL **: gtk_box_gadget_distribute: assertion 'size >= 0' failed in GtkSpinButton

[... 275 lines cut ...]

(pronterface.py:662): Gtk-CRITICAL **: gtk_box_gadget_distribute: assertion 'size >= 0' failed in GtkSpinButton

(pronterface.py:662): Gtk-WARNING **: Negative content width -12 (allocation 22, extents 17x17) while allocating gadget (node button, owner GtkButton)

(pronterface.py:662): Gtk-WARNING **: Negative content width -15 (allocation 19, extents 17x17) while allocating gadget (node button, owner GtkButton)

(pronterface.py:662): Gtk-CRITICAL **: gtk_box_gadget_distribute: assertion 'size >= 0' failed in GtkSpinButton

(pronterface.py:662): Gtk-CRITICAL **: gtk_box_gadget_distribute: assertion 'size >= 0' failed in GtkSpinButton

(pronterface.py:662): Gtk-CRITICAL **: gtk_box_gadget_distribute: assertion 'size >= 0' failed in GtkSpinButton

(pronterface.py:662): Gtk-CRITICAL **: gtk_box_gadget_distribute: assertion 'size >= 0' failed in GtkSpinButton

(pronterface.py:662): Gtk-CRITICAL **: gtk_box_gadget_distribute: assertion 'size >= 0' failed in GtkSpinButton

(pronterface.py:662): Gtk-CRITICAL **: gtk_box_gadget_distribute: assertion 'size >= 0' failed in GtkSpinButton

(pronterface.py:662): Gtk-WARNING **: Negative content width -12 (allocation 22, extents 17x17) while allocating gadget (node button, owner GtkButton)

(pronterface.py:662): Gtk-WARNING **: Negative content width -15 (allocation 19, extents 17x17) while allocating gadget (node button, owner GtkButton)

Scripts/run-pronterface-2.x.sh: line 5:   662 Segmentation fault      python3 pronterface.py -c /home/vanessa/.pronsolerc-2.x
```

rockstorm101, 2018-03-18:
Reproduced.

kliment, 2018-03-18:
Everything that required a reboot to take effect on py2/wxp3 causes a segfault on py3/wxp4

kliment, 2018-03-18:
Hmm, actually, I just tried to trigger this again so I could try to catch it in gdb and I can't get it to happen

kliment, 2018-03-19:
I think this is fixed with 61581e1ec2dbb32bd3e237cbbbdbe3d1e73aab8d - can you test?

hroncok, 2018-03-19:
I can open the temperature graph on my Fedora.

[image]

VanessaE, 2018-03-19:
The crash is mostly gone but if I open the large view and close it, the next time I click on the small graph, this error message appears in the console:

```
Exception ignored in: <bound method GraphWindow.__del__ of <printrun.gui.graph.GraphWindow object at 0x7f4293a68048>>
Traceback (most recent call last):
  File "/home/vanessa/RepRap/Printrun-2.x/printrun/gui/graph.py", line 43, in __del__
    super(GraphWindow,self).__del__(self)
AttributeError: 'super' object has no attribute '__del__'
Traceback (most recent call last):
  File "/home/vanessa/RepRap/Printrun-2.x/printrun/gui/graph.py", line 92, in show_graph_window
    self.window.Show()
AttributeError: 'NoneType' object has no attribute 'Show'
```
If I click again, the large view opens.

Everytime I open/close it, that error appears the next time I try to open it, effectively making it take two clicks to open.

I said "mostly gone" because if I then close Pronterface, it segfaults on the way down.

kliment, 2018-03-19:
Well spotted - can you try again with latest commit?

VanessaE, 2018-03-19:
That was with commit 89b28594940fb64e7c36d8840d3cd4bd70f69826

kliment, 2018-03-19:
Oh oops I forgot to push - try with 610f9ffbe7e3fe017b1595fc227000734b96a7e7

VanessaE, 2018-03-19:
That did it.

closed 2018-03-19 by VanessaE