kliment/Printrun #913: Negative content width

tobbelobb, 2018-03-14:
I had some of the errors as observed in #909's error message:
```
(pronterface.py:662): Gtk-WARNING **: Negative content width -12 (allocation 22, extents 17x17) while allocating gadget (node button, owner GtkButton)
(pronterface.py:662): Gtk-WARNING **: Negative content width -15 (allocation 19, extents 17x17) while allocating gadget (node button, owner GtkButton)
```
over and over.

I managed to get rid of them by enlargening the ">" and "+"-buttons.

[Commit here](https://github.com/tobbelobb/Printrun/commit/07d666513e86c7ad205d14b254605c77678f073e)

Just enlargening the buttons feels a bit dirty. Shouldn't they automatically resize? I found a discussion somewhere that suggested GTK+3.20 text resize happens at the wrong time: https://trac.wxwidgets.org/ticket/17585

tobbelobb, 2018-03-20:
The text field beside the send button looks squashed (gone) in your second screendump. Could that be the GtkEntry?

hroncok, 2018-03-20:
Probably.

kliment, 2018-03-23:
As this is now the only bug blocking the 2.x release I think it's time to make a new RC

hroncok, 2018-03-23:
On it.