kliment/Printrun #916: Messages printed twice in terminal

hroncok, 2018-03-21:
On 610f9ffbe7e3fe017b1595fc227000734b96a7e7

Right panel:

```
Connecting...
start
Printer is now online.
echo:Marlin 1.1.0.11
echo: Last Updated: 2016-04-27 12:00 | Author: (Aleph Objects Inc., LulzBot Mini)
...
```

Terminal:

```
Connecting...
Connecting...
Printer is now online.
Printer is now online.
```

hroncok, 2018-03-21:
Or load a file, if you don't have a printer around:

```
Loading file: /envy/dj/rostock_max/razor-holder.gcode
Loading file: /envy/dj/rostock_max/razor-holder.gcode
Loaded /envy/dj/rostock_max/razor-holder.gcode, 29565 lines
Loaded /envy/dj/rostock_max/razor-holder.gcode, 29565 lines
3041.45mm of filament used in this print
3041.45mm of filament used in this print
The print goes:
The print goes:
- from -31.91 mm to 31.91 mm in X and is 63.83 mm wide
- from -31.91 mm to 31.91 mm in X and is 63.83 mm wide
- from -15.41 mm to 11.25 mm in Y and is 26.66 mm deep
- from -15.41 mm to 11.25 mm in Y and is 26.66 mm deep
- from 0.00 mm to 18.05 mm in Z and is 18.05 mm high
- from 0.00 mm to 18.05 mm in Z and is 18.05 mm high
Estimated duration: 120 layers, 0:52:29
Estimated duration: 120 layers, 0:52:29
```

rockstorm101, 2018-03-21:
This is not a regression, it happened to me also with version 1.6.0.

kliment, 2018-03-22:
This should now be fixed in both 2.x and master - we were adding an extra console handler even though one had already been added, and not resetting the handlers when adding it.

rockstorm101, 2018-03-22:
I can confirm this is now solved, I see no more duplicated messages after 38e6c34.

kliment, 2018-03-22:
@hroncok can you test this works correctly for you now and close if so?

hroncok, 2018-03-22:
Confirmed.

closed 2018-03-22 by hroncok