kliment/Printrun #917: "Ok" messages while idle

djdelorie, 2018-03-21:
Printer: Rostock MAX V1, Repetier 0.92.9
Host: Fedora 26 x86-64

After connecting to my printer, the console window prints "Ok 0" every second or so.  During a print, it prints "Ok 1", "Ok 2", etc...  After the print, it prints "Ok 4294967295".  These messages cause other, more important messages, to scroll off the console.

Printer is now online.
Info:External Reset
zDistortionCorrection:1
Free RAM:2290
Autoretract:0
X:0.00 Y:0.00 Z:0.000 E:0.0000
SelectExtruder:0
FlowMultiply:100
ok 0
ok 0
ok 0
ok 0
ok 0
ok 0

rockstorm101, 2018-03-21:
Isn't this a duplicate of #813? This problem was supposed to be already dealt with.

djdelorie, 2018-03-21:
Turns out the problem still happens if you check the "monitor printer status" box, but that box has to be checked to enable the temperature graph, so... still a bug.  Also, the "Ok 1" etc messages are printed either way.

kliment, 2018-03-21:
This is actually an unrelated problem - repetier firmware has decided it will attach the line number it's okaying after the ok I guess - so we need to ignore an ok with a number after it, while we're currently only ignoring lines that end with "ok" or "wait".

kliment, 2018-03-21:
Just pushed a commit that should fix this, can you test?

djdelorie, 2018-03-21:
pronterface still prints the "ok 1" lines (along with "busy:processing" and "busy:heating" style messages)
(unless running ./pronterface.py in the source tree uses installed modules instead of those in the source tree)

kliment, 2018-03-22:
Which commit are you on?

djdelorie, 2018-03-22:
45417e31c51e56f599891b640f4a89355b679249

kliment, 2018-03-22:
Ok, I found what's going on. Can you try again with latest?

kliment, 2018-03-22:
Should also silence the busy: messages

djdelorie, 2018-03-22:
Seems to work.  THanks!

kliment, 2018-03-22:
Awesome.

closed 2018-03-22 by kliment

rockstorm101, 2018-03-22:
I still see tons of "echo:busy: processing" messages. I have the box called "Monitor printer status" checked. Should it be this way?
I use Marlin 1.1.8.

kliment, 2018-03-22:
I'll squelch those. Can you try with latest commit?

rockstorm101, 2018-03-22:
I'll try latest commit with the next print. In an hour or so. If you need any logs or something, let me know.

rockstorm101, 2018-03-22:
Now on 2e83002 I still see them. Also I see many "echo:Active Extruder: 0" or "echo:Active Extruder: 1" which might be expected I guess since it is a print with two colors.

kliment, 2018-03-22:
Do we want to hide those too? Can you come to IRC for a bit so we can discuss?