kliment/Printrun #918: Wish: move console

djdelorie, 2018-03-21:
I would like an option to place the console below the controls panel, instead of on the right, to leave more room for the plater.  This would be better for me as I have a large  monitor and use a "big" window for pronterface.

[image]

rockstorm101, 2018-03-22:
Hi @djdelorie  I believe this is already implemented. Could you try setting '_Settings_ > _Options_ > _User interface_ > _Interface mode_' to "Compact" instead of "Standard"? Let us know it that does what you need.

djdelorie, 2018-03-22:
SWEET!  I wish I had discovered that long ago...

rockstorm101, 2018-03-22:
I'll close the issue then since it seems solved.

closed 2018-03-22 by rockstorm101

hroncok, 2018-03-22:
For a reference:

[image]