kliment/Printrun #920: AttributeError: module 'sys' has no attribute 'frozen'

hroncok, 2018-03-23:
After c41e64333d5fec51e39a5ecd3a5caad1cfed4727

This is what is ee when I run pronterface from an updated Fedora package:

```
Traceback (most recent call last):
  File "/usr/bin/pronterface", line 62, in <module>
    app = PronterApp(False)
  File "/usr/lib64/python3.6/site-packages/printrun/pronterface.py", line 2350, in __init__
    self.mainwindow = PronterWindow(self)
  File "/usr/lib64/python3.6/site-packages/printrun/pronterface.py", line 184, in __init__
    self.SetIcon(wx.Icon(iconfile("pronterface.png"), wx.BITMAP_TYPE_PNG))
  File "/usr/lib64/python3.6/site-packages/printrun/utils.py", line 78, in iconfile
    return pixmapfile(filename)
  File "/usr/lib64/python3.6/site-packages/printrun/utils.py", line 109, in pixmapfile
    '/usr/share/pixmaps'])
  File "/usr/lib64/python3.6/site-packages/printrun/utils.py", line 100, in lookup_file
    if sys.frozen: prefixes+=[getattr(sys, "_MEIPASS", os.path.dirname(os.path.abspath(__file__))),]
AttributeError: module 'sys' has no attribute 'frozen'
```

Note it doesn't happen when I run pronterface form sources. Will investigate.

kliment, 2018-03-23:
Should be fixed with latest commit, can you test?

hroncok, 2018-03-23:
I had a bit different fix prepared (but basically the same), you are too fast :) Will apply the patch, rebuild the package and let you know.

The problem is when running form source, the code was never executed, because it returned sooner.

kliment, 2018-03-23:
Yeah, but this should be fairly robust.

hroncok, 2018-03-23:
Confirmed, fixed.

closed 2018-03-23 by hroncok