kliment/Printrun #921: Make pip installable wheels

hroncok, 2018-03-23:
I'll do this. After 2.0.0 final is released.

 * [ ] `setup.py` looks like a mess, let's clean it first
 * [ ] requirements are in `requirements.txt` only, move to (ordynamically load them from) setup.py `install_requires`
 * [ ] make wheels buildable, test them
 * [ ] possibly CI the wheels creation if @kliment  agrees :smile:

setherith, 2018-06-05:
I'm currently struggling like crazy on installing on Linux Mint 18 (x64). I went with the virtual environment approach as suggested but getting "Segmentation fault" when I start it. Please keep me updated on any changes to the install process for my OS, and if I can help in anyway I'll be glad to lend a hand :+1:

hroncok, 2020-07-16:
I'll try to look at this again this weekend.

hroncok, 2020-07-20:
I didn't make it, because I was offline all weekend, but it still si on my TODO.

hroncok, 2020-07-24:
I've opened https://github.com/kliment/Printrun/pull/1078 as step one.

If somebody could build and install the wheel on other platforms, that would be cool.

hroncok, 2020-08-02:
#1082 fixed the rest, but Ḯll close this once we have wheels on PyPI.

kliment, 2020-08-29:
Something is wrong with the wheels we're generating - when I install them with pip I can't run the application - it fails to find the ui image resources and crashes on startup. Pronsole seems to work. On macOS, it's even worse - the application fails to start, with the same error, but then instead of exiting just gets stuck forever until manually stopped. I'm not sure what's happening there, the images get installed to (venvpath)/share/pronterface/*.png but the application is looking for them in images/*.png

hroncok, 2020-08-29:
I'll have a look next week.

hroncok, 2020-08-29:
https://github.com/kliment/Printrun/blob/a4e0276fd9f33fbaad067048eaca0ff29bd500fc/printrun/utils.py#L80-L85

This (and other such constructs in the file) makes it so.

hroncok, 2020-08-29:
I believe we need to replace `/usr` (and `/usr/local`) with `sys.prefix` here. That should be venv agnostic.

hroncok, 2020-08-31:
@kliment please test from https://github.com/kliment/Printrun/pull/1101

kliment, 2020-09-05:
I tried the wheel that the CI built for #1101 and it fails at 'self.bg_bmp = wx.Image(imagefile(self.imagename), wx.BITMAP_TYPE_PNG).ConvertToBitmap()' in printrun/gui/xybuttons.py which expects to find a file in "images/control_xy.png" but the file is actually in sys.prefix+share/pronterface/control_xy.png so we're still losing a directory somewhere

closed 2020-09-06 by kliment

kliment, 2020-09-06:
We now have wheels on PyPI :)