kliment/Printrun #926: Part excluder somewhat broken (CPU usage, errors, kinda bad UI)

VanessaE, 2018-03-29:
When I launch the part excluder, the entire program becomes excruciatingly slow, pegging the CPU at 100% (one core).  If the print is still active, it starts stalling.

The excluder does eventually open, first showing me a black window, then showing me a plater, and tells me to draw rectangles over the areas to exclude, but gives no indication what I should do next, so I click "inject g-code" as a guess.  Doing so does one of three things:  Either it throws this error:

```
Traceback (most recent call last):
  File "/home/vanessa/RepRap/Printrun-2.x/printrun/gviz.py", line 88, in <lambda>
    self.Bind(wx.EVT_TOOL, lambda x: self.p.inject(), id = 6)
  File "/home/vanessa/RepRap/Printrun-2.x/printrun/gviz.py", line 208, in inject
    layer = self.layers[self.layerindex]
KeyError: 0
```

...or it pops up a dialog box asking me to enter some g-code, or it whites-out the view in the excluder window.

Regardless of which of those three things happen, any further actions in that window, other than closing it, cause the program to peg at 100% again for some time.  Even just resizing the window.  If the action is to open a menu (such as the little down arrow on the right), it hangs my window manager for a time.  That is, I can move my mouse around, but no actions anywhere are registered, not even window focus.

In any case, there's no clear way to indicate that I've finished boxing in the parts I want to exclude and that I'm ready to resume the print.   As a guess, I took the chance that correct action is just close the excluder, prime the nozzle, wipe it off, and click "Resume".  You should add a message somewhere to that effect.

kliment, 2018-03-29:
Yeah, the gcode tools (editor, excluder, plater) were made in a simpler age when gcode files were smaller and had distinct layers. There are severe performance issues there (and to a smaller degree in the STL plater) that need to be addressed, probably by the same method that was used to resolve the performance issue with initial gcode loading (cythonization). The gcode injector tool places gcode after a specific layer (this was originally contributed by someone who wanted to pause their print in a specific spot and get the nozzle out of the way so they could insert some captive nuts) but it seems to be trying to place some after a nonexistent layer. The UI issue is definitely there. So we have three separate issues here, and I'd appreciate if it could be split up into three separate issue reports as they have very different solutions.

closed 2018-03-29 by VanessaE

hroncok, 2018-03-29:
* https://github.com/kliment/Printrun/issues/927
 * https://github.com/kliment/Printrun/issues/928
 * https://github.com/kliment/Printrun/issues/929