kliment/Printrun #930: Occasionally can't exit

VanessaE, 2018-03-29:
Sometimes, very rarely, clicking File -> Exit or clicking the window's [X] button fails to actually quit, instead throwing this error:

```
Disconnecting from printer...
Are you sure you want to exit while printing?
(this will terminate the print).
y/n: Traceback (most recent call last):
  File "/home/vanessa/RepRap/Printrun-2.x/printrun/pronterface.py", line 350, in closewin
    self.do_exit("")
  File "/home/vanessa/RepRap/Printrun-2.x/printrun/pronsole.py", line 398, in do_exit
    if not self.confirm():
  File "/home/vanessa/RepRap/Printrun-2.x/printrun/pronsole.py", line 257, in confirm
    y_or_n = input("y/n: ")
EOFError: EOF when reading a line
```

No confirmation dialog is actually presented.

Workaround: click "disconnect" first, then exit the program.

#761 maybe be a reliable way to coax Pronterface into a state where this bug applies.

kliment, 2018-03-29:
This is much less rare than this report suggests - in fact it appears to happen every time you attempt to exit mid-print (when paused or when a print is ongoing). This should be fixed in  b48fe7bd582ea10cd126217aae2f67e036310100

kliment, 2018-03-29:
I'll leave this open until the 2.0.0 final release in case someone else encounters this issue while using the RC.

rockstorm101, 2023-08-13:
> I'll leave this open until the 2.0.0 final release

Issue is fixed and Printrun 2.0.0 is released. Closing this issue now.

closed 2023-08-13 by rockstorm101