kliment/Printrun #931: Pronterface panel background doesn't respect dark themes

lordofhyphens, 2018-03-31:
The background panel comes out hideous on a dark theme, and the high contrast washes everything else out.

[image]

For comparison, from Slic3r (which also uses wxWidgets):
[image]

kliment, 2018-03-31:
@hroncok can you look into this?

hroncok, 2018-03-31:
I have no idea how this works :(

@lordofhyphens Is this a 2.x regression? I.e. did it work correctly with 1.x?

hroncok, 2018-04-01:
1.x (yet with GTK3):

[image]

hroncok, 2018-04-01:
OK, I get when this is defined I guess:

```patch
diff --git a/printrun/gui/__init__.py b/printrun/gui/__init__.py
index e2d2d4d..fa75da2 100644
--- a/printrun/gui/__init__.py
+++ b/printrun/gui/__init__.py
@@ -132,13 +132,13 @@ class MainWindow(wx.Frame):
         return panel
 
     def registerPanel(self, panel, add_to_list = True):
-        panel.SetBackgroundColour(self.bgcolor)
+        panel.SetBackgroundColour("red")
         if add_to_list: self.panels.append(panel)
 
     def createTabbedGui(self):
         self.notesizer = wx.BoxSizer(wx.VERTICAL)
         self.notebook = wx.Notebook(self.panel)
-        self.notebook.SetBackgroundColour(self.bgcolor)
+        self.notebook.SetBackgroundColour("red")
         page1panel = self.newPanel(self.notebook)
         page2panel = self.newPanel(self.notebook)
         self.mainsizer_page1 = wx.BoxSizer(wx.VERTICAL)
```

[image]

hroncok, 2018-04-01:
We can utilize system theme background color, yet that would mean we need to rework all other colors as well:

[image]

[image]

hroncok, 2018-04-01:
Either we call it NOTABUG and as a workaround I recommedn you go to Options and change the background color:

[image]

Or we need to nuke the background color option and instead use the theme color everywhere.

hroncok, 2018-04-01:
I've created https://github.com/kliment/Printrun/pull/932 to workaround this a bit.

kliment, 2018-04-01:
That's a good solution - I think it's the best way to resolve this.

kliment, 2018-04-01:
Can you test your PR on OSX? I'm about to test on windows

kliment, 2018-04-01:
Hmm, not great on windows 
https://images.duckduckgo.com/iu/?u=https%3A%2F%2Fi1.ytimg.com%2Fvi%2FXxjFdG0_mes%2Fmqdefault.jpg&f=1

kliment, 2018-04-01:
Sorry, somehow github is failing to upload an image, I'll try again

kliment, 2018-04-01:
This is how it looks on windows with #932 :

[image]

hroncok, 2018-04-01:
Ok, let me hardcode whit for windows.

kliment, 2018-04-01:
Works on Windows now.

kliment, 2018-04-01:
I've merged #932, so I think this is resolved. Thanks @hroncok !

closed 2018-04-01 by kliment

lordofhyphens, 2018-04-01:
Did someone check with a dark theme on Windows? That is still a thing now and again (especially high contrast).

hroncok, 2018-04-01:
I don't know how to do themes on Windows. Yet users with dark themes can still set the background color in settings.

kliment, 2018-04-01:
Using system window color instead of system background color on windows gives correct results even with dark themes (but high contrast black theme has bad readability of the temp graph)